basler/pylon-ros-camera #138: Camera usage after ROS node

IgnacioEscorz, 2022-12-07:
Hello!
How can I properly stop the ROS camera node driver?
I've been succesfully able to create 3 different nodes for my 3 cameras, able to see them at RVIZ and RQT independently.
The problem is to try to use the cameras outside ROS, after usinf the node driver. Is like they bind to the driver and are not accesible anymore unless unplugged and plugged back in. For example, after using them in ROS, stop everything ROS related, I cannot get images even trying to use PylonViewer.

Thank you in advance.

IgnacioEscorz, 2022-12-19:
Extra Info, I found from the Pylon Software that after running the ROS driver, the camera stays with "Software trigger enabled on". The issue remains, how to properly shutdown a Pylon ROS camera driver. Thank you.

FrancoisPicardDTI, 2023-03-08:
Hello @IgnacioEscorz 
So I am not sure what you want to achieve here. When you say "properly shutdown", what do you mean exactly? What would be the behavior you would expect?

closed 2023-04-24 by FrancoisPicardDTI