basler/pylon-ros-camera #140: ace2 cameras requires targetBrightness in rosservice setBrightness as float in between [0,1]

sroscher, 2022-12-16:
Hello,
I'm using an ace2 camera (not an old ace) . According to BASLER support, this newer camera requires now a float in between 0 and 1 for the target_brightness in function:
bool setBrightness(const int& target_brightness, int& reached_brightness, const bool& exposure_auto,const bool& gain_auto);
in [pylon_camera_node.h](https://github.com/basler/pylon-ros-camera/blob/master/pylon_camera/include/pylon_camera/pylon_camera_node.h)

However, the current version of the ros code wants an integer. Could you please fix this?
______________________________________________________________________________________________________________________________________
The specific camera, I use, is: a2A2590-22gcBAS
when using the current rosservice setBrightness in combination with an ace2 camera and an integer value, this happens:
~ $ rosservice call /pylon_camera_node/set_brightness 120 true false true
reached_brightness: -997042304
reached_exposure_time: 4000.0
reached_gain_value: 0.5000000596046448
success: False
~ $ rosservice call /pylon_camera_node/set_brightness 120 true false true
reached_brightness: -1001849984
reached_exposure_time: 4000.0
reached_gain_value: 0.5000000596046448
success: False
~ $ rosservice call /pylon_camera_node/set_brightness 120 true false true
reached_brightness: -973736896
reached_exposure_time: 4000.0
reached_gain_value: 0.5000000596046448
success: False
~ $ rosservice call /pylon_camera_node/set_brightness 120 true false true
reached_brightness: -993516864
reached_exposure_time: 4000.0
reached_gain_value: 0.5000000596046448
success: False

Alex-Beh, 2023-03-23:
Hello, any update to this issue? I would like to set brightness to 0.3 (Value from pylonviewer) however the current ros wrapper only take float.

FrancoisPicardDTI, 2023-04-24:
Hello @Alex-Beh, @sroscher 
Can you test the "#140_set_brightness_float" ? Let me know if it solves your problem and do not hesitate to suggest changes.

sroscher, 2023-06-27:
Hello @FrancoisPicardDTI ,
the changes work so far, meaning the image is correctly adjusted. Thanks a lot for fixing it! The only thing that remains a bit weird to me is the console output of reached brightness. However, I also played around with a value above 1.0. Maybe that's the problem...:

ubuntu:~/camera/basler_camera/catkin_ws $ rosservice call /pylon_camera_node/set_brightness .5 true false true
reached_brightness: 83.0
reached_exposure_time: 5000.0
reached_gain_value: 0.0
success: False
ubuntu:~/camera/basler_camera/catkin_ws $ rosservice call /pylon_camera_node/set_brightness 1.5 true false true
reached_brightness: -1029930048.0
reached_exposure_time: 5000.0
reached_gain_value: 0.5000000596046448
success: False
ubuntu:~/camera/basler_camera/catkin_ws $ rosservice call /pylon_camera_node/set_brightness 1.0 true false true
reached_brightness: -1162340480.0
reached_exposure_time: 5000.0
reached_gain_value: 0.5000000596046448
success: False
ubuntu:~/camera/basler_camera/catkin_ws $ rosservice call /pylon_camera_node/set_brightness 0.1 true false true
reached_brightness: 78.0
reached_exposure_time: 5000.0
reached_gain_value: 0.0
success: False

When I find time, I will have a look at it, myself!