basler/pylon-ros-camera #141: Reduce Camera bandwidth a2A1920-51gcPRO

TRA55A, 2023-01-02:
Hi all,

I'm trying to run 6 cameras(a2A1920-51gcPRO) in parallel(I launch 6 times the pylon node with different device IDs).
I found out that even on 2.5Gbps ethernet (PCI-E card and supported POE switch), I get a lot of latency and a low frame rate.
I set the scale option to lower the resolution(690x432 res) and limited the FPS to 20 but for some reason, it's not enough.
[a2A1920-51gcPRO_40259370_ycbcr.txt](https://github.com/basler/pylon-ros-camera/files/10331334/a2A1920-51gcPRO_40259370_ycbcr.txt)

1. When I try to change the camera configuration to support compression I get only noise, any idea why?
2. How do I set the camera to free-run in ROS? 
3. Why the driver doesn't support YCbCr422? When I try to set it in the default.yaml the driver decides I want to use mono8 instead

Thanks,
TR.

FrancoisPicardDTI, 2023-03-08:
Hi @TRA55A 
1. When the compression on the camera is enabled, images are compressed in the camera first, and need to be decompressed in pylon then. This is not implemented in ROS yet, however, you could chekc the pylon C++ SDK sample "Utility_ImageDecompressor.cpp" to see how this works.
2. Use the dedicated ROS service to set the trigger mode to false.
3. It is indeed not implemented in the current pylon driver. But please feel free to submit a pull request if you're working on this issue!

m-binev, 2023-04-25:
@TRA55A Hi, regarding  YCbCr422 - there is no appropriate mapping format in ROS for that. So, either use Bayer or RGB8, please. I believe, the pylon ROS-driver specification is covering that. Regards

m-binev, 2023-04-25:
@TRA55A Hi again, regarding menaging the bandwidth of six cameras via a switch, I would recommend first using the pylon GigE Configurator tool to optimize the complete setup and then the pylon Bandwidth Manager to optimize the camera settings. Both tools are to be found in pylon Viewer > Tools. Please use pylon 7.2 or higher.

closed 2023-12-08 by FrancoisPicardDTI