basler/pylon-ros-camera #144: Missing diagnostic_updater.hpp file while bulilding workspace

rows-of-houses, 2023-01-17:
I have an issue while building my workspace:

`Starting >>> pylon_ros2_camera_component
--- stderr: pylon_ros2_camera_component                             
In file included from /home/user/cam_ws/src/pylon_ros2_camera/pylon_ros2_camera_component/src/pylon_ros2_camera_node.cpp:31:
/home/user/cam_ws/src/pylon_ros2_camera/pylon_ros2_camera_component/include/pylon_ros2_camera_node.hpp:82:10: fatal error: diagnostic_updater/diagnostic_updater.hpp: No such file or directory
   82 | #include <diagnostic_updater/diagnostic_updater.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
`
How can I deal with it?

FrancoisPicardDTI, 2023-02-21:
Hi @rows-of-houses 
You're definitely right and I have recently updated the documentation. Sorry for that.
You have to install it manually.
If you're using galactic: sudo apt install ros-galactic-diagnostic-updater
On humble : sudo apt install ros-humble-diagnostic-updater
Does it solve your problem?

FrancoisPicardDTI, 2023-03-08:
I am closing the issue. Please reopen it if it is not solved on your side.

closed 2023-03-08 by FrancoisPicardDTI