basler/pylon-ros-camera #145: rusrun Error: 3792764924

kaiiwoo, 2023-02-07:
Hi, successfully built workspace, but I have an error when I run `rosrun pylon_camera pylon_camera_node`.

The original error message is below.

[image]

FrancoisPicardDTI, 2023-02-21:
Hello @kaiiwoo 
I am not managing to reproduce the error. Did you manage to solve it?
I would say that the error message is coming from `pylon_camera_base.hpp:`
```
template <typename CameraTrait>
bool PylonCameraImpl<CameraTrait>::grab(Pylon::CGrabResultPtr& grab_result)
```
Line 527: `ROS_ERROR_STREAM("Error: " << grab_result->GetErrorCode() << " " << grab_result->GetErrorDescription());`
Can you confirm that please?

Can you try as well to set a user id to the camera?

FrancoisPicardDTI, 2023-03-08:
I am closing the issue. Please reopen it if the problem is not solved on your side.

closed 2023-03-08 by FrancoisPicardDTI