basler/pylon-ros-camera #150: use UserSet1 in rosrun image_view image_vi image:=/pylon_camera_node/image_raw  The video too slow

Nanmei, 2023-03-14:
this is pylon view setting:
[image]
[image]
[image]
![2023-03-14 10-31-45 的屏幕截图](https://user-images.githubusercontent.com/80303463/224885085-a05e3fc1-4d03-493b-b6d0-8
[image]
1891acfeb5e.png)
this is the launch and yaml:

![3](http
[image]
s://user-images.githubusercontent.com/80303463/224885340-3f281ad0-5944-4bdd-ba53-32979cf430cd.png)

first I save and load "UserSet2" in  pylon view,and in the launch I set startup_user_set : UserSet2, then roslacunch pylon_camera_node and rosrun image_view. When I move camera ,and 1 second later,image_view can show current picture,
but in the pylon view, the video is so fast.
https://user-images.githubusercontent.com/80303463/224888951-9e4fa271-cab1-4f35-8d69-ff60fb578c11.mp4

when I chang UserSet2 to CurrentSetting, image_view can show the video is so fast like pylon view show. 

https://user-images.githubusercontent.com/80303463/224889251-f9ac11c5-1a9c-480e-b6a0-713701296bf9.mp4

I  don't konw why , set UserSet2 can make image_view video too slow ,thanks, waiting for you reply.

FrancoisPicardDTI, 2023-04-24:
Hello @Nanmei 
image_view is a ROS tool so maybe it is coming from there and not from the ROS driver. In your application, did you write a client retrieving the images from the driver and displaying them? If it is slow then the probem may come from the driver. Otherwise, it may be a ROS related one.

Nanmei, 2023-09-21:
thanks for your reply,I add opencv code in pylon_camera_node.cpp. Opencv video show img_raw_msg_, but i watched the opencv video .it still slow .so I cant't know why   "  <arg name="startup_user_set" default="CurrentSetting" /> " set CurrentSetting  can work.  Set UserSet1 it work slow. I set frame_rate: 100, the video delay 1 second.  I think it is not  image_view problem. I need your help. thanks.
[image]
[image]

closed 2024-02-22 by FrancoisPicardDTI