basler/pylon-ros-camera #151: Add trigger source, mode, and selector access to startup configuration rather than via service

Lux-Liam, 2023-03-15:
Hi there,

I'm currently working on an application requiring timing synchronization with an external control system. The simplest and stablest way for us to do this is via hardware trigger. I understand that this node provides services to set these parameters, but it would be better if they could be configured via the launch file/config files.

If this is already supported, could you please indicate that this is so and provide instructions on the correct way to do so? For instance, if this is already as simple as adding the parameter to the config file, fantastic.

If the functionality is not supported, perhaps we could collaborate on making it possible while still respecting the way the code has been written.

Thank you, looking forward to talking about it.

Liam

FrancoisPicardDTI, 2023-04-21:
Hello @Lux-Liam 
Of course ! You're more than welcome to suggest a pull request where all the parameters you need can be specified in the launch file/config files. Have you already started the development?

rows-of-houses, 2023-05-04:
There is a way to call pylon's services via launch files, like this:
`<node pkg="rosservice" type="rosservice" name="reverse_x" args="call /camera/right/set_reverse_x true"/>`

Lux-Liam, 2023-05-04:
> There is a way to call pylon's services via launch files, like this: `<node pkg="rosservice" type="rosservice" name="set_line_mode_r" args="call /camera/right/set_line_mode 0"/>` It is important for different cameras to have different service names, like `set_line_mode_r` (right) or `set_line_mode_r` (left)

That's extremely helpful and solves the problem of configuring my system in a way that can be launched with a single command. Thank you for pointing this out.

Lux-Liam, 2023-05-04:
> Hello @Lux-Liam Of course ! You're more than welcome to suggest a pull request where all the parameters you need can be specified in the launch file/config files. Have you already started the development?

Hey @FrancoisPicardDTI ,

I've had implementing this in the node myself at the bottom of the backlog for a while as this has not been the bottleneck to a full implementation of our system. I think that between @rows-of-houses comment above and making clever use of the configuration parameters currently available and the implementation of loading user sets I can get all the way there. I will keep a note to open a PR here once I have some time to come back and implement it properly! Thanks again all.

closed 2023-05-04 by Lux-Liam