basler/pylon-ros-camera #152: using multiple basler camera

kaiiwoo, 2023-03-22:
Hi, thank you for opening nice open source.

How can I launch / run the code when I use multiple devices (2, 3. ,,, same camera)?

pengboyemo, 2023-04-13:
you can do like this,  add group in pylon_camera_node.launch.
<group ns="01">
    <arg name="config_file" default="$(find pylon_camera)/config/01.yaml" />
    <node name="$(arg node_name)" pkg="pylon_camera" type="pylon_camera_node" output="screen"
          respawn="$(arg respawn)" launch-prefix="$(arg launch_prefix)">
        <rosparam command="load" file="$(arg config_file)" />
        <param name="gige/mtu_size" value="$(arg mtu_size)"/>
        <param name="startup_user_set" value="$(arg startup_user_set)"/>
        <param name="enable_status_publisher" value="$(arg enable_status_publisher)"/>
        <param name="enable_current_params_publisher" value="$(arg enable_current_params_publisher)"/>
    </node>
</group>

<group ns="02">
    <arg name="config_file" default="$(find pylon_camera)/config/02.yaml" />
    <node name="$(arg node_name)" pkg="pylon_camera" type="pylon_camera_node" output="screen"
          respawn="$(arg respawn)" launch-prefix="$(arg launch_prefix)">
        <rosparam command="load" file="$(arg config_file)" />
        <param name="gige/mtu_size" value="$(arg mtu_size)"/>
        <param name="startup_user_set" value="$(arg startup_user_set)"/>
        <param name="enable_status_publisher" value="$(arg enable_status_publisher)"/>
        <param name="enable_current_params_publisher" value="$(arg enable_current_params_publisher)"/>
    </node>
</group>

FrancoisPicardDTI, 2023-04-21:
Hello @kaiiwoo and @pengboyemo 
Thanks for the input @pengboyemo. It is indeed one way to go. I am assuming that it is then solving your issue @kaiiwoo ?

closed 2023-06-14 by FrancoisPicardDTI