basler/pylon-ros-camera #154: error: 'ChunkSelector_AutoBrightnessStatus' is not a member of 'Basler_UniversalCameraParam:ChunkSelectorEnums'

deuknyeonkim, 2023-03-24:
Hi, i am novice in Linux OS......
i have a problems while catkin_make procedures like below.....
[image]

what can i do for fix the problems....

my pc version as like below.
Discription : Ubuntu 18.04.6 LTS
Codename : bionic
rosversion -d : Melodic

and hwinfo of camera is like below.
[image]

thank you and have a nice day~

xiaowhite-22, 2023-04-13:
@deuknyeonkim  i also had the same problem, please how did you solve it.

FrancoisPicardDTI, 2023-04-21:
Hello @deuknyeonkim and @xiaowhite-22 
I think you did not install the pylon SDK. I strongly suggest for you to look at the driver documentation and to look at all the tutorials ROS provide for. It is not an issue related to the driver itself, I am closing the issue.

closed 2023-04-21 by FrancoisPicardDTI