basler/pylon-ros-camera #156: Difference between ROS1 wrapper and ROS2 wrapper

Alex-Beh, 2023-04-26:
Hi, I am trying to use the ROS2 Humble wrapper and do hardware synchronization between two camera a2A2600-64umBAS - Basler ace 2. I am initially referring to this [issue](https://github.com/basler/pylon-ros-camera/issues/125). But I realized the parameters from [ROS1 default.yaml](https://github.com/basler/pylon-ros-camera/blob/master/pylon_camera/config/default.yaml#L36) is different from [ROS2 default.yaml](https://github.com/basler/pylon-ros-camera/blob/humble/pylon_ros2_camera_wrapper/config/default.yaml).

For example, `trigger_timeout` & `grab_timeout` are missing in ROS2 wrapper. So I would like to ask how much the difference between the ROS1 wrapper & ROS2 wrapper.

FrancoisPicardDTI, 2023-04-26:
Hello @Alex-Beh 
Regarding the differences between the frameworks ROS1 and ROS2, I suggest you have a look at the official documentation. Go through the tutorials if you don't know them.
As for the differences between the 2 drivers, there aren't that many. The 2 drivers offer the same functionalities and features. Parameters from the ROS1 driver have been either removed or renamed depending on their usage. ´trigger_timeout´ and grab_timeout´ are not missing. Maybe you're refering to the ´set_trigger_timeout´ and ´set_grab_timeout services´?

Alex-Beh, 2023-04-26:
Sorry for the misunderstanding. What I asking is the difference of ROS1 driver of pylon-ros-camera & ROS2 driver of pylon-ros-camera.

Yes I saw the services and it called the functions respectively. But I wonder is it possible to set it from parameters instead of manually calling the services? Lifecycle node seem not an option here also.

FrancoisPicardDTI, 2023-04-26:
Yes, you're right. It is not possible for now to access these parameters through the parameter server, only by these services. You can setup these parameters at startup through the dedicated yaml file and access them afterwards through the services. There's no plan for now to do it otherwise but it is definitely on our todo list for standardized implementation.

closed 2023-05-11 by Alex-Beh