basler/pylon-ros-camera #157: RISC-V version of pylon?

asimonov, 2023-05-04:
Hello, I see pylon SDK is available for ARM and x86 architectures.

Is there a version for RISC-V?
Or do you know if it is going to be released?

m-binev, 2023-05-05:
@asimonov Hi, no, there is no dedicated pylon version for RISC-V yet and nothing is planed in short-term time. Can you share some more details on the reasons for using it, the applications that would be involved, and in combination with what camera interfaces you would like to use it? Why not using ARM or x86? Regards

asimonov, 2023-05-09:
we are looking to try GigE cameras on RISC-V board as this may be our target hardware for some projects.
i am aware we can try Aravis, but was wondering if Basler would start supporting RISC-V...

closed 2023-05-30 by FrancoisPicardDTI