basler/pylon-ros-camera #160: [humble] Can't find pylon/PylonIncludes.h

bertaveira, 2023-05-14:
I am getting an error trying to build the `humble` branch. I have installed the pylon software suite and even tested with the pylon viewer.

However when I try to build I get this error:
`pylon_ros2_camera_component/src/tools/ip_auto_config.cpp:29:10: fatal error: pylon/PylonIncludes.h: No such file or directory`

Any ideas what I am doing wrong?

I installed with the Debian package downloaded from the Basler website. Version 7.3 on both ARM and x86 ubuntu 22.04

bertaveira, 2023-05-16:
Yeah I had to move to 7.x because the 6.x has no ARM Debian package. I find it weird how in the `humble` branch README it says that we should use 7.2 or later

whoutman, 2023-05-16:
Unfortunately I bumped into this issue as well.

It seems that the cmake/FindPylon.cmake is not properly included in CMakeLists.txt, as I was able to compile when commenting out the if-statement around the include of this cmake-file, see https://github.com/basler/pylon-ros-camera/blob/humble/pylon_ros2_camera_component/CMakeLists.txt#L39

With this if-statement, no output was generated for the messages in the lines below (I commented those out, see https://github.com/basler/pylon-ros-camera/blob/humble/pylon_ros2_camera_component/CMakeLists.txt#L41-L43), while the following output was reported without the if-statement:
```
->    Pylon include directories: /opt/pylon/include
->    Pylon library directories: /opt/pylon/lib
->    Pylon libraries: pylonbase;pylonutility;GenApi_gcc_v3_1_Basler_pylon;GCBase_gcc_v3_1_Basler_pylon
```

Did I forget to run a specific script? Running e.g. 
`source /opt/pylon/bin/pylon-setup-env.sh /opt/pylon`
does not seem to do the job.

As a reference:
pylon version 7.3.0 (downloaded deb package installer) with a x86 (64 Bit) architecture

alblinks, 2023-05-25:
I am experiencing the same error. Downgrading to pylon 6.3 worked.

Alex-Beh, 2023-06-22:
Facing the same issue and the solution by @whoutman works for me. Somehow the cmake cannot find the library but the Pylon_FOUND is set to 1. So the if case for the FindPylon.cmake has been skipped.

I have modified the if case to the following also and managed to compiled the packages with pylon 7.3 [👍](https://emojipedia.org/thumbs-up/)
```
if (NOT ${Pylon_FOUND} OR "$ENV{Pylon_INCLUDE_DIRS}" STREQUAL "")
    include("${CMAKE_CURRENT_SOURCE_DIR}/cmake/FindPylon.cmake")
endif()
```

Ekanshh, 2023-08-03:
> Facing the same issue and the solution by @whoutman works for me. Somehow the cmake cannot find the library but the `Pylon_FOUND` is set to 1. So the if case for the FindPylon.cmake has been skipped.
> 
> I have modified the if case to the following also and managed to compiled the packages with pylon 7.3 [+1](https://emojipedia.org/thumbs-up/)
> 
> ```
> if (NOT ${Pylon_FOUND} OR "$ENV{Pylon_INCLUDE_DIRS}" STREQUAL "")
>     include("${CMAKE_CURRENT_SOURCE_DIR}/cmake/FindPylon.cmake")
> endif()
> ```

Worked for me. :)

argolomb, 2023-08-07:
> Facing the same issue and the solution by @whoutman works for me. Somehow the cmake cannot find the library but the `Pylon_FOUND` is set to 1. So the if case for the FindPylon.cmake has been [...]

Work for me too.

nachovizzo, 2023-08-16:
@FrancoisPicardDTI, any plan on fixing this? It would be ideal to be able to build the ROS 2 wrapper without having to patch the build system.

On the README.md it says that Pylon SDK 7.2+ should be installed but this conflicts with the way of finding the package. The provided solution by @Alex-Beh works so seems to be easy to include in the ros2 branch.

What do you think? I'm happy to open a PR to fix this

ndahn, 2023-09-13:
Commenting out the if statement also worked on ROS noetic!

FrancoisPicardDTI, 2023-09-19:
Hello
I integrated the fixes suggested by @nachovizzo, PR #179.
Does it solve the issue?
Regards,
Francois

closed 2023-11-01 by FrancoisPicardDTI