basler/pylon-ros-camera #162: Feature request: camera emulator for the ros driver

whoutman, 2023-05-17:
Today, I was experimenting with the ros2 driver and the camera emulator.  When running
```
export PYLON_CAMEMU=1
ros2 launch pylon_ros2_camera_wrapper pylon_ros2_camera.launch.py
```
I got the following output
```
[INFO] [launch]: All log files can be found below /home/noblean/.ros/log/2023-05-17-08-40-41-638473-whoutman-34134
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [pylon_ros2_camera_wrapper-1]: process started with pid [34135]
[pylon_ros2_camera_wrapper-1] 1684312841.899847022 [basler.pylon.ros2.pylon_ros2_camera_parameter] [WARN] Autoflash: 0, line2: 1 , line3: 1 
[pylon_ros2_camera_wrapper-1] 1684312841.899906217 [basler.pylon.ros2.pylon_ros2_camera_parameter] [INFO] No Device User ID set -> Will connect the first available camera device
[pylon_ros2_camera_wrapper-1] 1684312842.208029227 [basler.pylon.ros2.pylon_ros2_camera] [INFO] Found camera device! Device Model: Emulation with Device User Id: 
[pylon_ros2_camera_wrapper-1] 1684312842.208082206 [basler.pylon.ros2.pylon_ros2_camera] [ERROR] The detected camera type is: BaslerCamEmu. Only 'BaslerUsb' and 'BaslerGigE' types are supported by this driver for now!
```

Hence, I observed that the camera emulator is not supported (for the ros driver). Would it be possible to include this feature?

SMA2016a, 2023-06-04:
PYLON_CAMEMU isn't implamented yet.

m-binev, 2024-02-06:
Hi @whoutman , as previously commented, the Camera Emulation transport layer is not supported yet (unlike USB, GigE).  Because the camera emulation behaves a little different than a real camera, I am not really optimistic it makes sense to implement it here at all. That is, its main purpose is the facilitate the typical code writing in case that a physical camera isn't available. Because ROS expects some preconditions to be met though, which the camera emulation would probably not meet, I assume this approach would make no sense here. 
Thus, I would suggest to go for using a standard Basler USB or GigE camera instead. Cheers

galou, 2024-05-16:
Cf. https://github.com/galou/pylon-ros-camera/tree/emulated_camera_humble

galou, 2024-05-16:
Cf. also https://github.com/galou/pylon-ros-camera/tree/emulated_camera based on *my* `master` (with other pending changes).