basler/pylon-ros-camera #164: Hardware trigger

mohamedezzatt, 2023-05-31:
I have an ace2 basler camera and i try to use the hardware trigger feature on camera line 1 but it gave me an error "Trigger device reset , pylon camera isn't ready". 
Due to the documentation , i have added those lines in applyCamSpecificStartupSettings function in the driver : 
            cam_->TriggerSelector.SetValue(Basler_UniversalCameraParams::TriggerSelector_FrameStart);
            cam_->TriggerMode.SetValue(Basler_UniversalCameraParams::TriggerMode_On);
            cam_->TriggerSource.SetValue(Basler_UniversalCameraParams::TriggerSource_Line1);
            cam_->TriggerActivation.SetValue(Basler_UniversalCameraParams::TriggerActivation_RisingEdge);

Any help about what could be the problem ?

SMA2016a, 2023-06-04:
I think this case can be closed? right?

FrancoisPicardDTI, 2023-06-14:
@mohamedezzatt Hello
Did you manage to solve your issue?
I am closing though as it is not related to the pylon ros driver. But do not hesitate to contact Basler's support if you still have th eproblem.

closed 2023-06-14 by FrancoisPicardDTI