basler/pylon-ros-camera #167: LineFormat_OpenDrain not included in setLineMode conditions

alireza-moayyedi, 2023-07-11:
I am using the a2A4504-18umPRO camera and I want to set the line mode to output. This is doable through the pylon viewer and I have already tested everything in my system and it works fine. However, when I try to use the availabe ROS 2 (humble) service to set the line mode it throws the following error:

**Error : the selected line number don't have change line mode feature**

Taking a deeper look I have noticed that only the TTL and LVTTL formats are allowed whereas in pylon viewer I see the OpenDrain format for this camera. Can OpenDrain also be added to the allowed conditions?

```
template <typename CameraTraitT>
std::string PylonROS2CameraImpl<CameraTraitT>::setLineMode(const int& value)
{
    try
    {   if ( (cam_->LineFormat.GetValue() == LineFormatEnums::LineFormat_TTL) || (cam_->LineFormat.GetValue() == LineFormatEnums::LineFormat_LVTTL) )
        {
            if (value == 0)
            {
                cam_->LineMode.SetValue(LineModeEnums::LineMode_Input);
            }
            else if (value == 1)
            {
                cam_->LineMode.SetValue(LineModeEnums::LineMode_Output);
            }
            else 
            {
                return "Error: unknown value";
            }
        }
        else 
        {
            RCLCPP_ERROR_STREAM(LOGGER_BASE, "Error : the selected line number don't have change line mode feature");
            return "Error : the selected line number dose not have change line mode feature";
        }
    }
    catch ( const GenICam::GenericException &e )
    {
        RCLCPP_ERROR_STREAM(LOGGER_BASE, "An exception while setting the line mode occurred:" << e.GetDescription());
        return e.GetDescription(); 
    }
    return "done";
}
```

alireza-moayyedi, 2023-07-13:
I have proposed a solution here:
https://github.com/basler/pylon-ros-camera/pull/168

FrancoisPicardDTI, 2023-12-22:
Hello @alireza-moayyedi 
I answered you on the PR. Let me know if it can be a way to solve your issue.

closed 2023-12-22 by FrancoisPicardDTI