basler/pylon-ros-camera #173: Config setting regarding camera_info_url

purple-mesh, 2023-07-24:
* ROS version and distro: ROS 2 humble (using the `humble` branch in the repo)
* Parameter in `pylon-ros-camera/pylon_ros2_camera_wrapper/config/default.yaml`

It mentions that:
```
    #  The CameraInfo URL (Uniform Resource Locator) where the optional intrinsic
    #  camera calibration parameters are stored. This URL string will be parsed
    #  from the ROS-CameraInfoManager:
    camera_info_url: "file:///path/to/my/file.yaml"
```

But I ended up having to tweak the `src/pylon-ros-camera/pylon_ros2_camera_wrapper/launch/pylon_ros2_camera.launch.py` launch script to include this as a LaunchConfiguration.

I think I am missing something here, since I assumed that no further tweaks would be required to load in the camera intrinsic matrix?

FrancoisPicardDTI, 2023-09-19:
Hello @purple-mesh 
Can you then attached the tweaked launch file? Did you move forward with this issue?
Regards,
Francois

purple-mesh, 2023-09-20:
Hi @FrancoisPicardDTI 

The updated version of the launch file which I use now is [available here](https://gist.github.com/purple-mesh/18fed7018b882b4a66e0cc48e87f1268)

No more issues after changing to use this.

But let me know if there was another way this was meant to be used.

FrancoisPicardDTI, 2023-12-08:
Hi @purple-mesh 
Trying to reproduce the issue here, without success for now.

* In my .yaml file:
`camera_info_url: "file:///ros2/temp/basler_usb_with_xxmm_lens_calibrationdata_2010_01_01/ost.yaml"`
* When I am starting the driver, in my terminal:
`[pylon_ros2_camera_wrapper-1] 1702047125.006242135 [my_camera.pylon_ros2_camera_node] [INFO] camera calibration URL: file:///ros2/temp/basler_usb_with_xxmm_lens_calibrationdata_2010_01_01/ost.yaml`

Do you see this message in your teminal without your hack?

purple-mesh, 2023-12-21:
Hi @FrancoisPicardDTI thanks for looking into this, sorry I haven't had a chance to get back to testing this, but will aim to test this again in Jan 2024.

FrancoisPicardDTI, 2024-02-22:
Hi @purple-mesh 
Did you have the chance to test on your side?

jordigc2, 2024-03-18:
Hello,

Are there any updates regarding this?
I am facing the same issue using ROS2 Humble.

purple-mesh, 2024-04-23:
Apologies @FrancoisPicardDTI for the belated response.
I have since been re-assigned to another project for now, so no longer have access to the original test-rig and computer(s).

@jordigc2 I presume you have the same issue, and were you able to fix it by updating the launch file / launch script?

FrancoisPicardDTI, 2024-10-31:
I am closing the issue for now, as there's no activity.

closed 2024-10-31 by FrancoisPicardDTI