basler/pylon-ros-camera #176: No tf camera frame neither camera_info are published

bach05, 2023-08-02:
Hello, 

I am running pylon-ros-camera to manage 2 Basler ACE cameras. I run under ROS Noetic (Ubuntu 20.04). I have calibrated the cameras and I can visualize rectified images (i.e. image_rect topics on rqt image_view), but camera_info and tf frames are not published by the camera node. 

These are my configuration and launch files.

**Configuration**
```
#The tf frame under which the images were published
camera_frame: basler0_frame

#  The DeviceUserID of the camera. If empty, the first camera found in the
#  device list will be used
device_user_id: "basler0"

#  The CameraInfo URL (Uniform Resource Locator) where the optional intrinsic
#  camera calibration parameters are stored. This URL string will be parsed
#  from the ROS-CameraInfoManager:
#  http://docs.ros.org/api/camera_info_manager/html/classcamera__info__manager_
#  1_1CameraInfoManager.html#details
camera_info_url: "file:///home/iaslab/orbecc_ROS1_ws/src/pylon-ros-camera/pylon_camera/config/calibration/calibrationdata_0/ost.yaml"

#  The encoding of the pixels -- channel meaning, ordering, size
#  taken from the list of strings in include/sensor_msgs/image_encodings.h
#  The supported encodings are 'mono8', 'bgr8', 'rgb8', 'bayer_bggr8',
#  'bayer_gbrg8' and 'bayer_rggb8'
#  Default values are 'mono8' and 'rgb8'
# image_encoding: "mono8"

#  Binning factor to get downsampled images. It refers here to any camera
#  setting which combines rectangular neighborhoods of pixels into larger
#  "super-pixels." It reduces the resolution of the output image to
#  (width / binning_x) x (height / binning_y).
#  The default values binning_x = binning_y = 0 are considered the same
#  as binning_x = binning_y = 1 (no subsampling).
# binning_x: 1
# binning_y: 1

#  The desired publisher frame rate if listening to the topics.
#  This parameter can only be set once at startup
#  Calling the GrabImages-Action can result in a higher framerate
frame_rate: 10.0

trigger_timeout: 5000  # in ms
grab_timeout: 500  # in ms

# Camera grab strategy 
# 0 : GrabStrategy_OneByOne
# 1 : GrabStrategy_LatestImageOnly
# 2 : GrabStrategy_LatestImages
grab_strategy : 0
```
**Launch pylon camera**
```
<?xml version="1.0"?>
<launch>

    <arg name="respawn" default="false" />
    <arg name="debug" default="false" />
    <arg name="node_name" default="pylon_camera_node" />
    <arg name="mtu_size" default="1500" />
    <arg name="startup_user_set" default="CurrentSetting" />
    <arg name="enable_status_publisher" default="true" />
    <arg name="enable_current_params_publisher" default="true" />
    <arg name="config_file" default="$(find pylon_camera)/config/default.yaml" />

    <arg unless="$(arg debug)" name="launch_prefix" value="" />
    <arg     if="$(arg debug)" name="launch_prefix" value="gdb -ex run --args" />

    <node name="$(arg node_name)" pkg="pylon_camera" type="pylon_camera_node" output="screen"
          respawn="$(arg respawn)" launch-prefix="$(arg launch_prefix)">
        <rosparam command="load" file="$(arg config_file)" />
        <param name="gige/mtu_size" value="$(arg mtu_size)"/>
        <param name="startup_user_set" value="$(arg startup_user_set)"/>
        <param name="enable_status_publisher" value="$(arg enable_status_publisher)"/>
        <param name="enable_current_params_publisher" value="$(arg enable_current_params_publisher)"/>
    </node>

</launch>

```
**Launch all cameras**
```
<launch>
    <!-- Launching femto_mega.launch -->
    <include file="$(find orbbec_camera)/launch/femto_mega.launch" />

    <!-- Launching pylon_camera_node.launch -->
    <include file="$(find pylon_camera)/launch/pylon_camera_node.launch" > 
        <arg name="config_file"  value="$(find pylon_camera)/config/basler0.yaml" />
        <arg name="node_name"  value="basler0" />
    </include>

        <!-- Launching pylon_camera_node.launch -->
    <include file="$(find pylon_camera)/launch/pylon_camera_node.launch" > 
        <arg name="config_file"  value="$(find pylon_camera)/config/basler1.yaml" />
        <arg name="node_name" value="basler1" />
    </include>

    <!-- Call rosservice to change image encoding for pylon_camera_node -->
    <node pkg="rosservice" type="rosservice" name="call_set_image_encoding0" args="call /basler0/set_image_encoding 'bgr8'" />
    <node pkg="rosservice" type="rosservice" name="call_set_image_encoding1" args="call /basler1/set_image_encoding 'bgr8'" />

</launch>
```

BruttherJOE, 2023-08-04:
Hi @bach05, I am currently facing the same issue with 2 basler ace cameras and on Noetic as well. Let us update on this thread if we have any breakthroughs

bach05, 2023-08-04:
My hypothesis it's we need a common parent frame (e.g. a fixed world frame) for both cameras. Otherwise tf is not able to manage both cameras. In fact if you run onyl ona camera, the tf will be published. 

I am going to calibrate extrinsics of both cameras wrt a fixed point to validate this hypothesis.

BruttherJOE, 2023-08-07:
I used the ros stereo cameracalibrator.py to calibrate the intrinsics and extrinsics.

How are you able to get tf with only one camera? did you do monocular calibration?

My rqt graph looks like this :
[image]

and I am receiving images on rect too. I republished camera_info for `stereo/left` and `stereo/right` from another topic that provides it, `/stereo/leica_basler_left/pylon_camera_node/camera_info` and that solves the problem of `camera_info` but image frame does not match?

[image]

I am new to stereo vision and I am quite possibly doing everything wrongly

bach05, 2023-08-07:
How are you able to get tf with only one camera? did you do monocular calibration?

> I used the ros stereo cameracalibrator.py to calibrate the intrinsics and extrinsics.
> 
> How are you able to get tf with only one camera? did you do monocular calibration?

Yes, you can do monocular calibration with a single camera. 

> and I am receiving images on rect too. I republished camera_info for `stereo/left` and `stereo/right` from another topic that provides it, `/stereo/leica_basler_left/pylon_camera_node/camera_info` and that solves the problem of `camera_info` but image frame does not match?

It's a bit hard to debug. There is a mismatch between the camera frame (in camera info header) and the image frame (in the image topic). Probably you should check the nodes that are publishing those information to fix the message headers in order to have the same frame. If you need to visualize the images only, you can just use the Image tool on Rviz.

FrancoisPicardDTI, 2023-09-19:
Hello @bach05 and @BruttherJOE 
Thank you for looking at this issue. Did you manage to implement some fixes at the end?
Best regards,
Francois

BruttherJOE, 2023-09-25:
hello @bach05 @FrancoisPicardDTI, 

I believe I did manage to fix my headers, and I was able to get camera_info and tf_frames from my calibrated files. However, the rectified images looked a bit wrong and was zoomed in wrongly possibly due to my calibration (I am using some custom stereo setup which goes through multiple lenses). Also, I was not able to get a pointcloud, perhaps due to this reason. Perhaps I do not know enough to debug these problems, but I calibrated my cameras in another way with matlab in the end as we can choose other (objectively better?) calibration patterns for our needs and converting / importing the parameters into opencv. The idea of using ros still interests me, so I will take a look at this in my free time.

Best Regards,
Joe

closed 2023-11-01 by FrancoisPicardDTI