basler/pylon-ros-camera #177: Potential Bug in setTriggerSource

Alex-Beh, 2023-08-07:
I am using [a2A2600-64umBAS](https://docs.baslerweb.com/a2a2600-64umbas#a2a2600-64umbas) and it only provide Line1, Line2 & Line 3.

However from the source code [setTriggerSource](https://github.com/basler/pylon-ros-camera/blob/621f0d4580e784cd5c3fd77f152f1c98745b0f69/pylon_ros2_camera_component/include/internal/impl/pylon_ros2_camera_base.hpp#L1958) it only provide Line1, Line3 & Line4. Is it a bug or a different design for different camera?

Relevat part:

1. [CurrentParams.msg](https://github.com/basler/pylon-ros-camera/blob/621f0d4580e784cd5c3fd77f152f1c98745b0f69/pylon_ros2_camera_interfaces/msg/CurrentParams.msg#L19)
2. [getTriggerSource](https://github.com/basler/pylon-ros-camera/blob/621f0d4580e784cd5c3fd77f152f1c98745b0f69/pylon_ros2_camera_component/include/internal/impl/pylon_ros2_camera_base.hpp#L2002C40-L2002C56)

Alex-Beh, 2023-08-29:
For my case, there are two GPIO pins available to me and I am using Line 2.

FrancoisPicardDTI, 2023-10-17:
Hi @Alex-Beh 
The current implementation is due to old camera design, and you're absolutely right, the new cameras imply a new implementation. I fixed it in the humble version:
0 -> Trigger Software / 1 -> Line 1 / 2 -> Line 2 / 3 -> Line 3 / 4 -> Line 4 / 5 -> Action 1 / 6 -> Periodic signal 1
Let me know if it solves your problem.

Alex-Beh, 2023-10-17:
Yes it did. Thanks for supporting this project.

closed 2023-10-17 by Alex-Beh