basler/pylon-ros-camera #180: [pylon_camera_node-2] process has died after launching pylon_camera_node

yangfei4, 2023-08-29:
Hi, I'm unable to run `roslaunch pylon_camera pylon_camera_node.launch`. I got the below error:

`yangfei@hostnamectlset-hostnamebigbertha:~/ur5e_collab_ws$ roslaunch pylon_camera pylon_camera_node.launch
... logging to /home/yangfei/.ros/log/fe786ba8-4693-11ee-8667-a8a1596807d8/roslaunch-hostnamectlset-hostnamebigbertha-126325.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hostnamectlset-hostnamebigbertha:45237/

SUMMARY
========

PARAMETERS
 * /pylon_camera_node/camera_frame: pylon_camera
 * /pylon_camera_node/camera_info_url: 
 * /pylon_camera_node/device_user_id: acA5472-17uc
 * /pylon_camera_node/enable_current_params_publisher: True
 * /pylon_camera_node/enable_status_publisher: True
 * /pylon_camera_node/frame_rate: 5.0
 * /pylon_camera_node/gige/mtu_size: 1500
 * /pylon_camera_node/grab_strategy: 0
 * /pylon_camera_node/grab_timeout: 500
 * /pylon_camera_node/startup_user_set: CurrentSetting
 * /pylon_camera_node/trigger_timeout: 5000
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    pylon_camera_node (pylon_camera/pylon_camera_node)

auto-starting new master
process[master]: started with pid [126382]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fe786ba8-4693-11ee-8667-a8a1596807d8
process[rosout-1]: started with pid [126440]
started core service [/rosout]
process[pylon_camera_node-2]: started with pid [126447]
[ WARN] [1693331194.787094075]: Autoflash: 0, line2: 1 , line3: 1 
[ INFO] [1693331194.787450699]: Trying to connect the camera device with the following device user id: acA5472-17uc
[pylon_camera_node-2] process has died [pid 126447, exit code -11, cmd /home/yangfei/ur5e_collab_ws/devel/lib/pylon_camera/pylon_camera_node __name:=pylon_camera_node __log:=/home/yangfei/.ros/log/fe786ba8-4693-11ee-8667-a8a1596807d8/pylon_camera_node-2.log].
log file: /home/yangfei/.ros/log/fe786ba8-4693-11ee-8667-a8a1596807d8/pylon_camera_node-2*.log`

If I run `roslaunch pylon_camera pylon_camera_ip_configuration.launch,` it's saying no device found:

`yangfei@hostnamectlset-hostnamebigbertha:~/ur5e_collab_ws$ roslaunch pylon_camera pylon_camera_ip_configuration.launch
... logging to /home/yangfei/.ros/log/eb053726-4694-11ee-8667-a8a1596807d8/roslaunch-hostnamectlset-hostnamebigbertha-126652.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hostnamectlset-hostnamebigbertha:39353/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    pylon_camera_node (pylon_camera/IPConfigAuto)

auto-starting new master
process[master]: started with pid [126709]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to eb053726-4694-11ee-8667-a8a1596807d8
process[rosout-1]: started with pid [126767]
started core service [/rosout]
process[pylon_camera_node-2]: started with pid [126771]
 no device found, will try once more in 2 s
 no device found, will try once more in 2 s
 no device found, will try once more in 2 s
`

And when I tried to open _pylon_camera_node-2*.log_ to check the error, this file is just non-exsit or empty.  
Additionally, I checked the connection and device_user_id in pylon_Viewer, everything works fine.
I'm so confused by this issue. Anyone can help me out? Thanks!!

FrancoisPicardDTI, 2023-09-19:
Hello @yangfei4 
It may be due to the fact that the IP settings of the camera are not well configured. You need to be within the same IP range (computer and camera). Look at the Troubleshooting section in the README file. Hopefully it will solve your problem.
Regards,
Francois

FrancoisPicardDTI, 2023-11-01:
Hello @yangfei4 
Did you end up solving the issue, by any chance?

yangfei4, 2023-11-01:
> Hello @yangfei4 Did you end up solving the issue, by any chance?

Yes, I solved this by running everything in a docker container. I don't know why but I couldn't launch the camera node locally.

FrancoisPicardDTI, 2023-12-22:
I am closing the issue then. Thanks!

closed 2023-12-22 by FrancoisPicardDTI