basler/pylon-ros-camera #184: not able to run acA5472-17uc camera. Topics are not going beyond 6Hz

rahulsharma11, 2023-10-02:
Hi.

I used pylon ros driver for multiple basler cameras and got no issue.

Right now i purchased another camera acA5472-17uc (17fps, 20MP). When i started this then no matter i change any parameter, it keeps on saying-
 "Desired framerate 15.00 is higher than max possible. Will limit framerate to: 6.01 Hz"

and i am not able to get even 10 hz of output data.

I tried both ROS and ROS2 driver but same error.

I used image_encoding : rgb8 
autoexposure : on
autogain: on

Any suggestions?

m-binev, 2023-10-02:
@rahulsharma11 Hi, well, for RGB8 pixel format you cannot get more than about 6.3 fps, because the USB3 bandwidth limits for more. Top get higher frame rate, you should change to Bayer8 pixel format. 
To figure out such topics better, you can either use the pylon viewer or alternatively the Frame Rate Calculator: https://www.baslerweb.com/de-de/tools/frame-rate-calculator/
Cheers

rahulsharma11, 2023-10-02:
Hi @m-binev .

Thanks for the reply and link.
Yes it is giving 6fps for RGB8.
I dont't want to use Bayer8 as i need colour image to process.

Why they have given officially 17fps. 
I am using intel NUC i7 and that seems good processor.

They should have given 17fps but very extreme high end processor.

m-binev, 2023-10-02:
@rahulsharma11 Hi, it's not the CPU, but the USB3 bandwidth itself. 17fps is what you get when using Bayer8 - so it's correct. When using RGB8 the camera has to transfer 3x more data, which is too much for the USB3 interface.

satyajit-ink, 2023-10-06:
Hi @m-binev i am facing a similar issue, on pylon viewer i am able to get 17-18 fps on USB3, so it the viewer using bayer8 by default?

m-binev, 2023-10-13:
@satyajit-ink Hi, the pylon Viewer is not configuring the camera automatically, but it uses/takes the default/current camera settings. For most Basler cameras it holds true, that they are "waking up" with parameters being set up to deliver the maximum possible frame rate according to the specification. Once, you start changing parameters like Pixel Format, etc., that may influence and data throughput, you of course could get a different frame rate. This can also happen, if you increase the exposure time to a value that requires longer sensor exposure. I think, the Basler online product documentation can help a lot with such quaestions (select your camera model first): https://docs.baslerweb.com/