basler/pylon-ros-camera #185: Frame Transmission Delay

MaxBial, 2023-10-12:
The currently available functionalities in the ROS-Driver for the Transport Layer are:

```
(GigE only) GevSCPSPacketSize (Packet Size)
(GigE only) GevSCPD (Inter-Packet Delay)
(USB only) MaxTransferSize
```

This is enough for running a single camera - but we have a project where we are running up to 5 cameras. In such a case it is essential to have control over the functionality:

`(GigE only) GevSCFTD (Frame Transmission Delay)`

We cannot receive the images of all 5 cameras in the exact same moment, therefore there is the option to shift in time/receiving the 5 images with the functionality "Frame Transmission Delay".

Is there a possibility to incorporate it also in the ROS-Driver to use in the config file as "Packet Size" (MTU) and "Inter-Packet Delay"?

Best Regards,
Max

m-binev, 2023-10-13:
@MaxBial Hi, yes, you are absolutely right about the Frame Transmission Delay and its purpose in a multi-camera setups. As you have access to the full source code, why don't you add this parameter yourself and make a pull request? This would be much appreciated. Cheers
@FrancoisPicardDTI

FrancoisPicardDTI, 2023-10-17:
Hello @MaxBial 
Which version of the ROS driver are you running with?

MaxBial, 2023-10-19:
Hello @FrancoisPicardDTI 

ROS1 - so basiclly the main branch.

Best Regards,
Max

FrancoisPicardDTI, 2023-11-01:
Hello @MaxBial 
Please try the newest master branch. It includes now the frame transmission delay setting.
Let me know if it solves your problem.
Best regards,
Francois

closed 2024-01-31 by FrancoisPicardDTI