basler/pylon-ros-camera #186: Chunk Mode Timestamp

FlorianPfab, 2023-10-18:
Hello,

i am trying to get the Chunk Timestamp from my Basler acA1920-50gc camera with ROS2 humble. Somehow I get an error saying that my camera is not supporting the chunk feature. According to the Basler Documentation (https://docs.baslerweb.com/data-chunks), the Ace1 cameras do provide the ChunkTimestamp parameter.

Following procedure:
1. Launch the camera `ros2 launch pylon_ros2_camera_wrapper pylon_ros2_camera.launch.py`
2. Stop grabbing (to set the chunk mode) `ros2 service call /my_camera/pylon_ros2_camera_node/stop_grabbing std_srvs/srv/Trigger {}`
3. Set Chunk Mode active `ros2 service call /my_camera/pylon_ros2_camera_node/set_chunk_mode_active std_srvs/srv/SetBool data:\ true`
4. Set Chunk Selector to 29 (=Chunk Timestamp Enum) `ros2 service call /my_camera/pylon_ros2_camera_node/set_chunk_selector pylon_ros2_camera_interfaces/srv/SetIntegerValue value:\ 29`
5. Enable Chunk `ros2 service call /my_camera/pylon_ros2_camera_node/set_chunk_enable std_srvs/srv/SetBool data:\ true`
6. Start grabbing again `ros2 service call /my_camera/pylon_ros2_camera_node/start_grabbing std_srvs/srv/Trigger {}`
7. Get Chunk Timestamp `ros2 service call /my_camera/pylon_ros2_camera_node/get_chunk_timestamp pylon_ros2_camera_interfaces/srv/GetIntegerValue {}`

I would expect to receive the timestamp, since all of the conditions are set (when checking with the getter-services, everything set above is fine). But instead, i get following error:
`[pylon_ros2_camera_wrapper-1] 1697652455.332167369 [basler.pylon.ros2.pylon_ros2_camera_base] [ERROR] Error while trying to getting the Chunk Timestamp. The connected Camera not supporting this feature`

I tried recreating the problem in the pylon viewer and there everything works fine (Open the camera, go to the "Chunk Data Stream"-Menu, set chunk mode to active, select Timestamp, enable the chunk; Start grabbing in the pylon viewer, go to the "Chunk Data"-Menu and you can see the Chunk Timestamp getting in).

I would appreciate any help since i do not know how to pin down the root of the problem any more!

Thanks!

m-binev, 2023-10-18:
@FlorianPfab Hi, well, to be honest I am not sure that using chunks on ROS has been implemented yet and I am not sure how it is supposed to work at all. I believe that what you are doing on ROS is not comparable to what is happening in pylon viewer and in the pylon APIs at all. That is, in pylon (or in GigE Vision), you are getting the chucks out of the image and not from the camera (what I believe you are doing). Please have a look at the pylon C++ API and the sample showing how chunks are extraxted out of the image: https://docs.baslerweb.com/pylonapi/cpp/sample_code#grab_chunkimage 
As I'm not that big ROS expert, I cannot say if that is supposed to work in ROS like that or at all. 
Hope that helps. Cheers
@FrancoisPicardDTI FYI

FrancoisPicardDTI, 2023-12-22:
Hello @FlorianPfab 
Working on the issue here. I can reproduce the problem. I'll investigate it more deeply after the Christmas holydays.

FlorianPfab, 2024-02-06:
Hi @FrancoisPicardDTI,
thank you for looking into it, is there any progress yet? Or any guesses where it goes wrong?
Thanks!

FrancoisPicardDTI, 2024-02-07:
Hi @FlorianPfab 
Thank you for reahing out. Working currently on the issue, it may be due to the way the pylon API is used. Coming back to you as dast as possible.

FrancoisPicardDTI, 2024-02-12:
Hi @FlorianPfab 
It should be fixed. Pull the latest changes and let me know.

FrancoisPicardDTI, 2024-02-22:
Hi @FlorianPfab 
Did you have the chance to test the new driver version by any chance?

closed 2024-04-08 by FrancoisPicardDTI