basler/pylon-ros-camera #187: Blaze camera dependency

tonynajjar, 2023-10-19:
Recently the prerequisite of installing [pylon Supplementary Package for blaze](https://www.baslerweb.com/de/support/downloads/downloads-software/) was added to the humble branch, breaking the build for anyone without this prerequisite. 
Can it somehow be made as an optional prerequisite to not break the build for those not using that camera? Thanks

m-binev, 2023-10-19:
@FrancoisPicardDTI May you check that please. Making the prerequisite optional makes sense to me.

FrancoisPicardDTI, 2023-10-20:
Hello @tonynajjar 

We have decided to include the blaze functionalities within the ROS2 branches, in order to ease the maintenance process of these branches. So you have to follow the new installation guidelines to build the pylon driver, and install the supplementary package for Blaze provided by Basler.

Regards,
Francois

tonynajjar, 2023-10-20:
Alright that's your call however it would be nice to follow the best practice of not breaking the build on an old release and instead have the change lined up for the next release with some migration/release notes ([example](https://navigation.ros.org/migration/Iron.html)). Thank you

FrancoisPicardDTI, 2023-10-26:
You're absolutely right. Thank you for this suggestion.

closed 2023-10-26 by tonynajjar

FrancoisPicardDTI, 2023-10-26:
@tonynajjar do you recommend the use of tags through git?

tonynajjar, 2023-10-26:
Well it depends what you want to achieve, tags are just a way to mark a static point in the git history. To release this package to the ROS buildfarm, I believe you will need to create tags, see https://docs.ros.org/en/iron/How-To-Guides/Releasing/Releasing-a-Package.html for more info