basler/pylon-ros-camera #193: [Humble] Boost COMPONENTS libraries getting replaced by pylon_ros2_camera_component

alireza-moayyedi, 2023-12-08:
Hello,

I was trying to update the drivers and use the latest commit of the Humble branch in my project but I started having some compilation issue with my Boost library imports. After some debugging, I noticed that the Boost libraries, as specified in my cmake, were being replaced as the following:

my cmake:

```
find_package(Boost REQUIRED COMPONENTS system thread program_options)
message("These are my boost libraries after finding Boost package and before finding pylon_ros_camera_component: ${Boost_LIBRARIES}")
find_package(pylon_ros2_camera_component REQUIRED)
message("These are my boost libraries after finding pylon_ros_camera_component package: ${Boost_LIBRARIES}")
```

And I get the following while trying to build my package:
```
These are my boost libraries after finding Boost package and before finding pylon_ros_camera_component: Boost::system;Boost::thread;Boost::program_options
These are my boost libraries after finding pylon_ros_camera_component package: Boost::system;Boost::filesystem;Boost::date_time;Boost::iostreams;Boost::serialization
```
Of course the simple fix would be to put the Boost find package after the pylon_ros2_camera_component, but I was not sure if this is normal. It took me a while to figure it out and my hypothesis is that the pylon-supplementary-package-for-blaze is causing it.
I am not sure if this is the right place to flag this issue, but at least I hope this can be helpful for anyone else facing the same issue.

FrancoisPicardDTI, 2023-12-22:
Hello @alireza-moayyedi 
So your suggestion is just to switch the 2 find_package commands, am I right? Is that how you solved your problem?

alireza-moayyedi, 2023-12-22:
Hello @FrancoisPicardDTI 
That is indeed how I have solved my problem. But in my opinion that problem must not happen in the first place. I mean a package should not overwrite libraries that I specify for myself in my cmake. But somehow changes inside **find_package(pylon_ros2_camera_component REQUIRED)** are being propagated into my cmake.

FrancoisPicardDTI, 2024-02-08:
Hello @alireza-moayyedi 
Boost is not embedded into the pylon-supplementary-package-for-blaze, and not the pylon API either. So I guess it is not coming from the driver itself.