basler/pylon-ros-camera #194: No "running" status message when using specific device user id

landersson, 2023-12-11:
Hi, 

What's the reason for only publishing a "running" status message if `pylon_camera_parameter_set_.deviceUserID()` is empty?

https://github.com/basler/pylon-ros-camera/blob/e22aa7fd4b4215c83a576b3e987d51965245383a/pylon_ros2_camera_component/src/pylon_ros2_camera_node.cpp#L615

I'm trying to add support for plug-and-play autodetection of Basler cameras. I launch a ros2 driver instance for each camera, using the `device_user_id` parameter to make the driver wait for a particular camera to become available. But with the logic applied on the line linked above, I never get a "running" status messager, because `pylon_camera_parameter_set_.deviceUserID()` is not empty.

FrancoisPicardDTI, 2023-12-22:
Hello @landersson 
Maybe can you suggest some changes through a dedicated pull request?

FrancoisPicardDTI, 2024-02-22:
Hello @landersson 
Trying to reproduce the problem here.

What I did on my side:
- My launch file points to one given .yaml file where I have specified the user id of the camera I want to connect
- I am launching a ros2 drive instance, through this launch file, and it connects to the specified camera
- If I am unplugging it and then plugging it again, the driver connects to the camera without any problem
- If I am launching the instance with the camera unplugged, and then plug it, the driver connects to the camera

So in your setup:
- Where do you specify this device user id?
- How do you start the driver exactly?

FrancoisPicardDTI, 2024-10-31:
I am closing the issue for now, as there's no activity.

closed 2024-10-31 by FrancoisPicardDTI