basler/pylon-ros-camera #195: Driver installation problem

Heronal, 2023-12-19:
Hello ! 

I have an issue since few days. I was working with this pkg and it was perfectly working on my workspace. But I needed to clean it and then I removed the Basler driver and the pkg from my computer. Since then I can't execute this line : 

`sudo sh -c 'echo "yaml https://raw.githubusercontent.com/basler/pylon-ros-camera/master/pylon_camera/rosdep/pylon_sdk.yaml" > /etc/ros/rosdep/sources.list.d/30-pylon_camera.list' && rosdep update && rosdep install --from-paths . --ignore-src --rosdistro=$ROS_DISTRO -y`

I have this kind of error no matter the version of the driver (I tried from 5.0.12 to 7.4.0) : 
```
updated cache in /home/icam/.ros/rosdep/sources.cache
Found an existing Pylon installation in PYLON_ROOT but the version is too low
Found an existing Pylon installation in PYLON_ROOT but the version is too low
executing command [rosdep-source install https://raw.githubusercontent.com/magazino/pylon_camera/indigo-devel/rosdep/pylon_sdk.rdmanifest]
Found an existing Pylon installation in PYLON_ROOT but the version is too low
ERROR: installation failed:
source: installation script returned with error code
ERROR: the following rosdeps failed to install
  source: command [rosdep-source install https://raw.githubusercontent.com/magazino/pylon_camera/indigo-devel/rosdep/pylon_sdk.rdmanifest] failed
```
My guess is that it comes from the driver pylonSDK installation but I don't know what can cause this...

By the way, I am working with ROS1 on ubuntu 20.04.
Don't hesitate if you need more informations  ; )

Have a nice day

FrancoisPicardDTI, 2024-02-22:
Hello @Heronal 
Looks like more a rosdep problem. I am not sure what you cleaned. What do you get when you echo PYLON_ROOT? Did you reinstall pylon as well?

Heronal, 2024-02-23:
Hello, I did fix the problem, it was a kind of rosdep problem, i just removed entirely all thé files (because i had installed multiple versions of the driver) and install it again and it worked !

Thank you for your time

closed 2024-02-23 by FrancoisPicardDTI

lhhht, 2024-06-10:
@Heronal  May I ask which version of the driver you reinstalled to solve the issue?

lhhht, 2024-06-10:
I encounter the same error as you when I install pylon6, but there are no errors when I install pylon5.

Heronal, 2024-06-10:
Hello, i reinstalled the 7.4.0 version.

But in my case it was more because i had some files left in my computer from previous versions.