basler/pylon-ros-camera #196: AutofuntionROI Offset OutOfRangeException

francescoo99, 2023-12-29:
Hello,

I'm currently implementing the `AutofunctionROI` feature but encountering challenges setting the `AutoFunctionROIOffsetX` and `AutoFunctionROIOffsetY` parameters for the autofunction ROI. Here's a breakdown of my setup:

- Camera: Basler camera with a sensor supporting a maximum resolution of 1936x1216.
- Image ROI: Configured to a size of 1584x1200.
  - Maximum X offset: 352 (1936 - 1584)
  - Maximum Y offset: 16 (1216 - 1200)

```c
cam_->AutoFunctionROISelector.SetValue(AutoFunctionROISelector_ROI1);
cam_->AutoFunctionROIOffsetX.SetValue(x_offset_to_set)
cam_->AutoFunctionROIOffsetY.SetValue(y_offset_to_set)
cam_->AutoFunctionROIWidth.SetValue(width_to_set);
cam_->AutoFunctionROIHeight.SetValue(height_to_set);
cam_->AutoFunctionROIUseBrightness.SetValue(true);
```

Now, aiming to define an Autofunction ROI smaller than the Image ROI, here are the values I'm attempting to set:
- `x_offset_to_set` = 812
- `y_offset_to_set` = 404
- `width_to_set` = 288
- `height_to_set` = 288

[image]

When I try to do this, I get the following exception: 
`[basler.pylon.ros2.pylon_ros2_camera_base] [ERROR]  Value = 812 must be equal or smaller than Max = 352`  

My confusion arises from observing that the AutoFunction ROI's X offset has the same maximum value as the Image ROI. Despite intending to create a smaller Autofunction ROI that fits entirely within the Image ROI, I'm encountering identical constraints on the maximum offsets.

Attached is a visual representation to illustrate my point, acknowledging that the proportions may not be exact but serve to demonstrate the issue.

Thank you in advance

m-binev, 2024-01-15:
@francescoo99 Hi, I admit, I have not tried to figure out & to a full entent everything you've posted, but logically thinking, I'd first set the widht/height before setting offset x/y - otheriwise the camera cannot know it's dealing with a smaller ROI.
Hope that helps

francescoo99, 2024-01-23:
Hi @m-binev,  after conducting some tests, I can confirm that the issue was indeed that one. Thank you for the support!

FrancoisPicardDTI, 2024-01-31:
Hello @francescoo99 
You did reopen the issue. Can you tell why? Do you have an issue?

francescoo99, 2024-01-31:
No, just my mistake!

closed 2024-01-31 by francescoo99