basler/pylon-ros-camera #197: Frame rate not adjusting with new ROI

J-C-Haley, 2024-01-04:
Hello, I'm trying to use the SetROI service to reduce data load, and my expectation was that frame rate would rise to the maximum (as a function of exposure, allowed bandwidth, and set frame rate), as it does in the pylonviewer GUI. However, current behavior is to limit the frame rate on startup ([here](https://github.com/basler/pylon-ros-camera/blob/a7cad55f4a7c7a4e5a5a39810e5605ce344619d0/pylon_camera/src/pylon_camera/pylon_camera_node.cpp#L624C15-L624C50)), based on the startup settings (ROI/exp/framrate), and this overrides the frame_rate ROS parameter, even if the SetROI ([here](https://github.com/basler/pylon-ros-camera/blob/a7cad55f4a7c7a4e5a5a39810e5605ce344619d0/pylon_camera/src/pylon_camera/pylon_camera_node.cpp#L1220)) were to attempt to reset the frame rate the original parameter has been destroyed, so further dynamic changes have no rate increase effect. 

So, my question is, shall we either:
- have a separate parameter to store the intended max framerate (e.g. target_frame_rate), so we preserve the user intent, or,
- not override the ros frame_rate parameter with the startup max framerate, and simply report achieved rate in the terminal or possibly another param (achieved_frame_rate)

I'd prefer the latter, it's a quick fix I can contribute, but I'd like to hear input on if this would break anything.

Thank you!

J-C-Haley, 2024-01-04:
Ah, I see this is a bigger problem than I assumed, frame rate gets baked into the main ros::Rate spin on startup, which is not adjustable from within the pylon_camera_node. Either of these fixes would require transferring ownership of the rate into the node, so it can be adjusted...

FrancoisPicardDTI, 2024-01-31:
Hello @J-C-Haley 
I just added a section in the documentation.
There are several issues, and several solutions, to increase the frame rate.
If the camera image acquistion is triggered by sofware trigger (default setting), then it is not possible to get the maximum frame rate, because the image acquisition is sequentially triggered, which is not overlapping then.
Several possible solutions are mentionned and tested in issue [#21](https://github.com/basler/pylon-ros-camera/issues/21), [#28](https://github.com/basler/pylon-ros-camera/issues/28), [#29](https://github.com/basler/pylon-ros-camera/issues/29), [#81](https://github.com/basler/pylon-ros-camera/issues/81), [#116](https://github.com/basler/pylon-ros-camera/issues/116), [#147](https://github.com/basler/pylon-ros-camera/issues/147), and [#200](https://github.com/basler/pylon-ros-camera/issues/200). Please refer to them for more information and hopefully find a solution for your use caase.

amohiud2, 2024-06-21:
few things that helped me with getting better frame rate were:

1. In one computer there was a USB port issue. Everything gets half the speed. 
2. Disable the parameter publishing in the launch file.
3. Disable software trigger (that doubles the frame rate)
4. In my case I needed less exposure time, so when I used minimum exposure time, I was able to get better frame rate. It actually does check how much you can get based on your frame rate settings in the config file. If it cannot it will say that we can only get a particular frame rate.
5. There is also a parameter "fast mode" or "slow mode", it works with pylon viewer and gets me speeds up to 750 fps, vs 554 fps. However when I use the service call to set this set_sensor_mode param to 1, it does not work. It says node is not writable etc.

FrancoisPicardDTI, 2025-09-30:
Hello @amohiud2 @J-C-Haley 
From driver version 3.1.0, enabling free run is now possible. Please check out [the dedicated branch](https://github.com/basler/pylon-ros-camera/tree/jazzy_3.0.4_freerun) implementing this feature (only for Jazzy and beta version). Beware that even if the API stays unchanged, some default parameters have been changed, which may result in expected behaviors from the system and user end. In any case, maximum frame rate can be reached now.
Let me know if you have any feedback. I am closing the issue for now.