basler/pylon-ros-camera #198: multiple USB cameras retrieveresult delay

P1ucks, 2024-01-08:
I have a device equipped with 18 USB cameras, employing hardware triggers. Using the retrieveresult() function to capture an image from each camera in sequence . write a code to log the execution time of this function. I noticed that one or more cameras experience delays of several hundred milliseconds when executing this function, while the rest of the cameras remain stable within 10 milliseconds. When I switch cameras or interfaces, the issue persists with a different camera.

m-binev, 2024-01-15:
@P1ucks Hi, well, my first feeling is that I actually lack some/a lot of information about your setup and what you are exactly doing... That is why, I'll suggest some general checks first.
From the following document: https://www2.baslerweb.com/media/downloads/software/pylon_software/readme_17.txt 
go to Performance Optimization > USB Vision Cameras > try out all suggested measures. For sure increase the URBs size and the File Handle Limit though.
Of course, I hope you are not using long exposure times (also defined by the trigger signal), because long exposure means longer time))
Cheers

FrancoisPicardDTI, 2024-10-31:
I am closing the issue for now, as there's no activity.

closed 2024-10-31 by FrancoisPicardDTI