basler/pylon-ros-camera #199: Question about function PylonInitialize

Felix00643298, 2024-01-22:
Hey，
I am wondering if I have 2 basler cameras, can I call `PylonInitialize` tiwce in each individual basler camera object in 2 seperate threads?
Thanks.

Felix00643298, 2024-01-23:
> Hi @Felix00643298, well PylonInitialize() ist loading the complete pylon runtime environment for the given application and should be called ideally only once. For more information you may also refer to: https://docs.baslerweb.com/pylonapi/cpp/pylon_programmingguide#initializationuninitialization-of-the-pylon-runtime-library

In my ROS system, there are multiple Basler cameras, and I initialize each camera sequentially using ROS plugins. This is a multi-threaded system, and currently, I call PylonInitialize once in the constructor of each Basler class. However, it seems to make the program very unstable, and I'm unsure if what I'm doing is incorrect.
```C++
basler::basler()
{
    PylonInitialize();
}
basler::~basler()
{
    PylonTerminate();
}
```

FrancoisPicardDTI, 2024-02-22:
Hello @Felix00643298 
Do you use at all the ROS Basler driver in your setup? You can call one driver instance per camera.

Felix00643298, 2024-03-15:
> Hello @Felix00643298 Do you use at all the ROS Basler driver in your setup? You can call one driver instance per camera.

Thanks for replying me.
Do you mean using one dirver instance to init all basler camera and make sure **PylonInitialize** is only called once?

FrancoisPicardDTI, 2024-04-08:
Hello @Felix00643298 
You are posting here an issue that should be related to the ROS pylon driver: https://github.com/basler/pylon-ros-camera
Are you using this driver to operate your cameras? Otherwise, you need to contact Basler support I guess.

Looking at your issue, it should not be a problem to call in several instances PylonInitialize. Can you explain a bit more what you are trying to do ? and eventually provide with code examples so that your issue can be reproduced ? You are writing that your program is very unstable, can you be more specific and provide with error messages ?

FrancoisPicardDTI, 2024-10-31:
I am closing the issue for now, as there's no activity.

closed 2024-10-31 by FrancoisPicardDTI