basler/pylon-ros-camera #200: pylon_camera fps not reaching set frame rate of 100 hz

JaleesaHoule, 2024-01-23:
Hi there, 

I am using pylon_camera for a Basler a2A1920-160umBAS and am encountering the same issue as previously described in [#27](https://github.com/basler/pylon-ros-camera/issues/27). The same message is spit out `[ INFO] [1583442203.551151871]: Start image grabbing if node connects to topic with a frame_rate of: 100 Hz` when launching. No errors are given, but when checking the topic `rostopic hz /pylon_camera_node/image_raw`, the frame rate is maxing out between 42-56fps. 

I have tried the following:

- Adjusting set_trigger_selector, set_trigger_source, set_trigger_activation, and set_trigger_mode service commands (no change to fps)
- Lowering the exposure (increases the frame rate slightly)
- Changing grab_strategy (no change)
- Changing the trigger_timeout and grab_timeout (no change)
- Changing binning_x and binning_y (changes fps slightly)

I have also tried reverting to one of the older commits of ros-pylon-camera that were supposed to have fixed the issue (around Dec 15, 2020 according to #27). 

There are no issues running the camera at 100 - 150 fps in pylon. Is this a camera specific issue or is there any setting that may help to resolve this? 

Thanks in advance for your help.

JaleesaHoule, 2024-01-25:
Hi @m-binev, thank you for taking the time to respond. 

Please correct me if I am wrong, but I am under the impression that changing settings in Acquisition Mode in pylon (viewer) will not affect the camera settings used when running pylon_camera in ros. In order to get the same settings in ros, I have been using the list of service commands in the [pylon_camera documentation](https://github.com/basler/pylon-ros-camera/blob/master/pylon_camera/README.rst).

In pylon (viewer), letting the camera run (Acquisition Mode = Continuous + Trigger Mode = Off) gives the desired frame rate of 100Hz. Correspondingly, I have tried to run the pylon_camera node with the service commands 'set_trigger_mode: =false' and 'frame_rate:=100', but the resulting frame rate is maxing out around 52 Hz.

m-binev, 2024-01-25:
Hi @JaleesaHoule, when loading the ros driver, some/many camera settings get overwritten. Thus, you have to do the "right" configuration for your use case in ROS. 
I think, just setting the trigger mode to "off" won't be enough. 
I suggest you check some of the closed issues related to "low frame rate", e.g., #116 , #81

JaleesaHoule, 2024-01-26:
Hi @m-binev, it seems like the solution for both #116 and #81 was to change the 'enable_current_params_publisher' setting to false. Unfortunately, this only seems to improve performance from about 52 to 63 Hz. At this point, I have tried most of the parameters that are listed in the documentation, but have not been able to get to the desired 100 Hz frame rate. Again, I can get to 150+ Hz easily with pylon (viewer), so I do not believe this is a computer problem. Are there any other settings that can be disabled to potentially improve camera speeds? Could this be a compatibility issue with the version of ROS that I'm running (noetic)? 

These are the current parameters I have set: 

PARAMETERS
 * /pylon_camera_node/camera_frame: pylon_camera
 * /pylon_camera_node/camera_info_url: 
 * /pylon_camera_node/device_user_id: 
 * /pylon_camera_node/enable_current_params_publisher: False
 * /pylon_camera_node/enable_status_publisher: False **(I have tried having this set to True as well - doesn't change anything)**
 * /pylon_camera_node/exposure: 4000.0 **(I have played around with the exposure - lowering it to 20.0 increases performance by 2-3 Hz)**
 * /pylon_camera_node/frame_rate: 100.0
 * /pylon_camera_node/gige/mtu_size: 3000
 * /pylon_camera_node/grab_strategy: 0 **(I have tried all 3 grab strategies, no change in performance)**
 * /pylon_camera_node/set_trigger_mode: False
 * /pylon_camera_node/startup_user_set: CurrentSetting
 * /rosdistro: noetic
 * /rosversion: 1.16.0

fhwedel-hoe, 2024-02-14:
@JaleesaHoule Can you try if https://github.com/fhwedel-hoe/pylon_instant_camera gives you maximum FPS?

rahulAquila, 2024-03-21:
Have there been any resolutions to this? This is a critical issue within this library that is hindering us from using it in any of our applications, the instructions on read me are not sufficient and misleading as well.

fhwedel-hoe, 2024-03-21:
Hi @rahulAquila

Indeed, it is a bit hidden. The [readme](https://github.com/basler/pylon-ros-camera?tab=readme-ov-file#slow-frame-rate) to this project says "it is not possible to get the maximum frame rate, because the image acquisition is sequentially triggered".

You may use an external trigger source (I have no experience with that) or use a different implementation (such as mine, see above). There are other implementations as well. You surely can find one that fits your needs. :)

JaleesaHoule, 2024-03-21:
@fhwedel-hoe Thank you for working on a solution to this problem. 

The documentation states "If the camera image acquistion is triggered by sofware trigger (default setting), then it is not possible to get the maximum frame rate." However, the documentation does give instructions for changing the trigger mode so that it is not on a software trigger. Therefore, according to the documentation, it should theoretically be possible to get the maximum frame rate if the trigger settings are changed. If this is not in fact possible, the documentation should be updated. 

That said, @rahulAquila, I have not managed to achieve the desired frame rate using any of the suggested fixes above or in the other relevant issues. The max fps that I was able to achieve after playing around with the exposure and other settings was around 92 hz, and those results were inconsistent. This project has been shelved for the time being, so I have not yet tried @fhwedel-hoe's node to see if that may work for my needs, but will plan to give that a shot when I circle back to this project.

ne01230, 2024-10-30:
rosservice call /pylon_camera_node2440/set_trigger_mode "data: off"
close the software trigger, the frame rate reach the max frame rate as pylon viewer reached.

FrancoisPicardDTI, 2025-06-03:
Hello @JaleesaHoule @ne01230 @fhwedel-hoe @rahulAquila 
I have recently adjusted the documentation to reflect the driver behavior: https://github.com/basler/pylon-ros-camera/blob/jazzy/README.md#acquisition-mode-and-frame-rate
Let me know if it clarifies things.

FrancoisPicardDTI, 2025-09-30:
Hello @JaleesaHoule @ne01230 @fhwedel-hoe @rahulAquila 
From driver version 3.1.0, enabling free run is now possible. Please check out [the dedicated branch](https://github.com/basler/pylon-ros-camera/tree/jazzy_3.0.4_freerun) implementing this feature (only for Jazzy and beta version). Beware that even if the API stays unchanged, some default parameters have been changed, which may result in expected behaviors from the system and user end.
Please let me know if you have any feedback. I am closing this issue for now.

closed 2025-09-30 by FrancoisPicardDTI