basler/pylon-ros-camera #202: Camera Config help for short range imaging

jitaxis, 2024-01-25:
How do I set the blaze configuration so that it works well in short distances (up to 1.5 m). I also don't want auto features, I would like to set the exposure, gain and so on. I have been able to change the exposure and frame rate, but I haven't had much luck setting other parameters.
Also when I set the exposure low the quality of the image looks to drop.

Any suggestions are greatly appreciated.
Thank you.

jitaxis, 2024-01-30:
@FrancoisPicardDTI 
Do you have any thoughts?
I have realised in the readme that I can only set the FPS and exposure for the Blaze. I can change so much more in the viewer, but the main thing i would want to do is turn gamma correction off. Is there a way to do this in ROS?

timonehr, 2024-02-06:
You can parameterize the blaze camera via the service interface. 

A few examples for you:
```
root@VM:/home/test_ws# ros2 service call /my_blaze/pylon_ros2_camera_node/enable_hdr_mode std_srvs/srv/SetBool "{data : "true"}"
waiting for service to become available...
requester: making request: std_srvs.srv.SetBool_Request(data=True)

response:
std_srvs.srv.SetBool_Response(success=True, message='done')
```

```
root@VM:/home/test_ws# ros2 service call /my_blaze/pylon_ros2_camera_node/set_temporal_filter_strength pylon_ros2_camera_interfaces/srv/SetIntegerValue "{value : 100}"
requester: making request: pylon_ros2_camera_interfaces.srv.SetIntegerValue_Request(value=100)

response:
pylon_ros2_camera_interfaces.srv.SetIntegerValue_Response(success=True, message='done')
```

The `enable_gamma_correction` service for a blaze camera is currently missing.

FrancoisPicardDTI, 2025-02-19:
Hello @jitaxis 
What is the status of your issue? Did you try what @timonehr suggested at the end?

closed 2025-09-30 by FrancoisPicardDTI