basler/pylon-ros-camera #206: No available camera. Keep waiting and trying...

zarathacia, 2024-03-04:
I am running the pylon ros camera package on a Noetic docker image. the installation was done successfully. The camera is connected using a USB connection. lsusb gives the camera id. but when running the pylon camera node is unable to find the camera. 

[image]
I put the device user id in the default.yaml file like this: 

[image]

SMA2016a, 2024-04-03:
does lsusb see the camera?

zarathacia, 2024-04-05:
hello, yes lsusb sees the camera. it turns out the installed driver is not compatible with the camera i possess. I needed to install the version 6.
Thank you for the response.

closed 2024-04-05 by zarathacia