basler/pylon-ros-camera #221: Blaze depth map encoding

bracoe, 2024-05-30:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hello,

I want to use the Blaze together with the Ros2 [depth_image_proc](https://github.com/ros-perception/image_pipeline/tree/rolling/depth_image_proc/include/depth_image_proc) package, but they require either _16UC1_ encoding or _32FC1_. Is it possible for the _/my_camera/pylon_ros2_camera_node/blaze_depth_map_ topic to be published as one of these? "image_encoding (not for the blaze)" and my attempts suggest it isn't. But I'm hoping I missed something!

Alternatively, how would one convert from _mono16_ (or any other supported encoding) to _16UC1_? Or is the idea to go from the pointcloud to a depth image instead?

### Hardware setup description

Blaze-101

### Runtime information

```shell
Humble
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

SMA2016a, 2024-05-31:
please use CV_16UC1

bracoe, 2024-05-31:
Thanks for the reply. Could you please show me how to actually use your depthmap as a CV_16UC1? 

Just replacing the encoding string to a CV_16UC1 format results in a 10 times bigger pointcloud when running it through _depth_image_proc::PointCloudXyzNode_.  Using CV_bridge to convert the encoding results in the same issue.

Trying to set the wrapper to output _16UC1_ results in an error since it is not supported.

[image]

[image]

bracoe, 2024-06-03:
Just to clarify. I'm wondering how we get an actual unit of measurement from the depth map (meters or millimeters). To us right now it looks like a value between 0 to 255 based on the Blaze's min/max parameters. Is a custom node needed to convert this to measurement data instead of a mono representation of the depth? And are our assumptions about the depth map topic correct?

bracoe, 2024-06-07:
Hello,

We think we found the problem.

https://github.com/basler/pylon-ros-camera/blob/2577c67d60bf770541bc660bca0b82035a1a103a/pylon_ros2_camera_component/include/internal/impl/pylon_ros2_camera_blaze.hpp#L659-L660

Causes the depth map to be warped and does not give the expected behavior as an actual depth map. Switching it to

` double distance = pPoint->z * 0.152588; `

fixes the issues we've had. The constant should of course  be from camera.Scan3dCoordinateScale.GetValue() instead of hard coded.

What is the intention with using the radial distance instead of the actual distance? Especially in a ROS context.

FrancoisPicardDTI, 2024-06-18:
Hello @bracoe 
I changed the code. Please have a look at it. Thank you for your contribution.

closed 2024-06-18 by FrancoisPicardDTI

Srijal97, 2024-08-11:
Hello @FrancoisPicardDTI, so with this change the depth map reports depth in millimeters now, correct?