basler/pylon-ros-camera #222: Cannot access ethernet camera from within docker container

theateist, 2024-06-01:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I'm trying to connect to the camera from a docker container.

OS: Ubuntu 22.04.4 LTS
Pylon: pylon_7.5.0.15658-deb0_amd64

The camera is connected to ethernet interface `enp0s31f6` which is on static ip `192.168.2.10/255.255.255.0`. The camera is on static ip `192.168.2.11/255.255.255.0`

I built [pylon-ros-camera](https://github.com/basler/pylon-ros-camera?tab=readme-ov-file) and it successfully detected the camera on the host
[image]

But even though I successfully built the same package inside a docker container, it cannot detect the camera
[image]

**What I tried**

- To allow the container to see the interfaces on the host, I run the container with networking as docker run `--network=host -d myimage ....` And now if I run `ifconfig` inside the container, it shows me the same interfaces as on the host. See below `ifconfig` output on host and inside container. But, it still cannot detect the camera.
- I ran the container with extended privileges as `docker run --network=host --privileged -v /dev:/dev -d ....` But, it still cannot detect the camera.

What might be the problem?

### Hardware setup description

**IFCONFIG on host**

```
docker0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 172.17.255.255
        inet6 fe80::42:cff:fe29:3048  prefixlen 64  scopeid 0x20<link>
        ether 02:42:0c:29:30:48  txqueuelen 0  (Ethernet)
        RX packets 602988  bytes 31801390 (31.8 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 701650  bytes 5591761213 (5.5 GB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

enp0s31f6: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.2.10  netmask 255.255.255.0  broadcast 192.168.2.255
        inet6 fe80::ada5:60b0:e461:3007  prefixlen 64  scopeid 0x20<link>
        ether 9c:6b:00:07:61:79  txqueuelen 1000  (Ethernet)
        RX packets 10880082  bytes 16425539680 (16.4 GB)
        RX errors 9  dropped 0  overruns 0  frame 9
        TX packets 54503  bytes 3794035 (3.7 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 19  memory 0x72500000-72520000  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 240419  bytes 70007245 (70.0 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 240419  bytes 70007245 (70.0 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp4s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.19  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::40ef:b0b9:5215:18e7  prefixlen 64  scopeid 0x20<link>
        ether f0:b6:1e:a2:b6:92  txqueuelen 1000  (Ethernet)
        RX packets 6337099  bytes 8614290755 (8.6 GB)
        RX errors 0  dropped 6672  overruns 0  frame 0
        TX packets 1492973  bytes 542835548 (542.8 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

**IFCONFIG inside container**

```
root@docker-desktop:/home/basler_ws# ifconfig
br-aa73862743b3: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 172.20.0.1  netmask 255.255.0.0  broadcast 172.20.255.255
        inet6 fe80::42:18ff:fe68:6f5e  prefixlen 64  scopeid 0x20<link>
        ether 02:42:18:68:6f:5e  txqueuelen 0  (Ethernet)
        RX packets 12  bytes 699 (699.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 17  bytes 12202 (12.2 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

br-fb4a1a6ffdce: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.2.1  netmask 255.255.255.0  broadcast 192.168.2.255
        inet6 fe80::42:12ff:feaf:abcc  prefixlen 64  scopeid 0x20<link>
        ether 02:42:12:af:ab:cc  txqueuelen 0  (Ethernet)
        RX packets 26  bytes 4052 (4.0 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 68  bytes 13912 (13.9 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

docker0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 172.17.255.255
        ether 02:42:be:4c:46:18  txqueuelen 0  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.65.9  netmask 255.255.255.0  broadcast 192.168.65.255
        inet6 fe80::5054:ff:fe12:3456  prefixlen 64  scopeid 0x20<link>
        ether 52:54:00:12:34:56  txqueuelen 1000  (Ethernet)
        RX packets 19853  bytes 2562859 (2.5 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 18195  bytes 7005442 (7.0 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 140  bytes 18520 (18.5 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 140  bytes 18520 (18.5 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

services1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.65.6  netmask 255.255.255.255  broadcast 0.0.0.0
        inet6 fe80::3c19:b1ff:fe38:ffcb  prefixlen 64  scopeid 0x20<link>
        ether 3e:19:b1:38:ff:cb  txqueuelen 0  (Ethernet)
        RX packets 46  bytes 3709 (3.7 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 83  bytes 15611 (15.6 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

veth150498a: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::30b3:9eff:fe24:7a7a  prefixlen 64  scopeid 0x20<link>
        ether 32:b3:9e:24:7a:7a  txqueuelen 0  (Ethernet)
        RX packets 26  bytes 4416 (4.4 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 109  bytes 19474 (19.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

### Runtime information

```shell
ROS_DISTRO = humble
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

SMA2016a, 2024-06-25:
Sending the ping from docker to 192.168.2.10 works fine?
Sending the ping from docker to camera works fine?

if not will you create a tcp dump in the docker while starting your application in the docker

FrancoisPicardDTI, 2024-10-31:
Hello @theateist 
Did you manage to solve your issue at the end?

Pandananana, 2024-11-13:
I have the same issue. I can connect from the host and i can also ping the camera from the docker container, but i can't connect