basler/pylon-ros-camera #223: Not able to change Line2 and Line3 mode with camera Ace2

Typos91, 2024-06-04:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I use an Ace 2 camera a2A5320-34gmBAS, and I am trying to connect a light which will be controlled through the camera. 

For this, I need to be able to activate or deactivates the Lines 2 and 3. I want to change their mode to **Output**, and then choose 2 different sources for each line: **UserOutput1** and **UserOutput2** for instance, so that I can control them "manually".
I referred to the pylon's [documentation](https://docs.baslerweb.com/user-output-value) to understand the process. 

Using the ROS2 driver, I understand that I have to use the *services* to execute all different steps:
- Select the Line I want to modify: **int 1** for Line 2
- Change the line mode to **Output**: **int 1** for Output
- Set the Line source to the right source: **int 2** for UserOutput1. Here, I can only change the source to UserOutput1 according to the ros2 driver's documentation, however in the Pylon's documentation it states that UserOutput1,2 and 3 are available for my camera.
- Activate or deactivate the chosen source: **bool**

This operation has worked using the Pylon Viewer, and the different UserOutput sources appear

However, when I try to change the mode of Line2, I get this error:

```bash
[pylon_ros2_camera_wrapper-1] ... [basler.pylon.ros2.pylon_ros2_gige_camera] [ERROR] Error : the selected line number does not have change line mode
```

I tried to change the mode of line 3 but I get the same error.

I searched inside the driver, it seems to me that the error might come from the file **pylon_ros2_camera_gige.hpp**, in the method **setLineMode**. 

Here is an extract of my code:

```c++
SetLineSelector_client = this->create_client<SetIntegerSrv>("my_camera/pylon_ros2_camera_node/set_line_selector");
SetLineMode_client = this->create_client<SetIntegerSrv>("my_camera/pylon_ros2_camera_node/set_line_mode");

 /*============================================Selecting line 2 ====================================================*/
 auto line_selector_request = std::make_shared<SetIntegerSrv::Request>();
   line_selector_request->value = 1;
   RCLCPP_INFO(this->get_logger(), "Waiting for service SetLineSelector to be avalailable");
   while(!SetLineSelector_client->wait_for_service(1s))
   {
       if (!rclcpp::ok())
       {
           RCLCPP_ERROR(this->get_logger(), "Interrupted while waiting for the service . Exiting");
           return;
       }
       RCLCPP_INFO(this->get_logger(), "service not available, waiting again...");
   }

   RCLCPP_INFO(this->get_logger(), "Service available. Sending request now");
   auto line_selector_response = SetLineSelector_client->async_send_request(line_selector_request);
   if (rclcpp::spin_until_future_complete(this->get_node_base_interface(), line_selector_response) ==
rclcpp::FutureReturnCode::SUCCESS)
   {
       if(line_selector_response.get()->success)
       {
           RCLCPP_INFO(this->get_logger(), "Line 2 selected");
       }
       else
       {
           RCLCPP_ERROR(this->get_logger(), "Failed to select Line 2");
           return;
       }
   }
   else
   {
       RCLCPP_ERROR(this->get_logger(), "Failed to connect to service 'Line Selector'");
   }
   /*============================================Changing line mode to output ====================================================*/

   RCLCPP_INFO(this->get_logger(), "Set Line Mode to 'output', waiting for server to be available");
   while(!SetLineMode_client->wait_for_service(1s))
   {
       if (!rclcpp::ok())
       {
           RCLCPP_ERROR(this->get_logger(), "Interrupted while waiting for the service . Exiting");
           return;
       }
       RCLCPP_INFO(this->get_logger(), "service not available, waiting again...");
   }
   auto line_mode_request = std::make_shared<SetIntegerSrv::Request>();
   line_mode_request->value = 1; // 1 for output
   auto line_mode_response = SetLineMode_client->async_send_request(line_mode_request);
   if (rclcpp::spin_until_future_complete(this->get_node_base_interface(), line_mode_response) ==
rclcpp::FutureReturnCode::SUCCESS)
   {
       if (line_mode_response.get()->success)
       {
           RCLCPP_INFO(this->get_logger(), "Line mode set to output");
       }
       else
       {
           RCLCPP_ERROR(this->get_logger(), "Failed to set line mode to output");
           return;
       }
   }

```

I also tried to do this after having stopped the grabbing, since I saw in the method **setLineSourceCallback** from **pylon_ros2_camera_node.cpp**, that errors could occur if I try to change line's mode while grabbing images with the camera.

Then, I have two questions:
- Is there another step to do before that I might have missed, to be able to change the line mode ?
- When the line mode is changed to output, how can I select different **UserOutput** sources for each line using the services ?

### Hardware setup description

PC :
- CPU architecture x86_64
- Operating system : Ubuntu 22.04
- RAM 8 GiB

Interfaces used to connect the camera:
- Ace 2 camera a2A5320-34gmBAS
- Power source : 20 VDC
- no switches/hubs
- Cable : Basler Power-I/O Cable, M8 6p/open, P; 5m
- Light connected: LXE300 Direct Connect Linear Light
- Pin used for the connection to the light bar : Line2 -->change mode of the light (Continuous or strobe), Line3 -->NPN Trigger 

### Runtime information

```shell
pylon info: 7.4.0.38864 64-Bit

----------------------------------------

pylon Setup info: 7.4.0.14900 

----------------------------------------

pylon Application info: 2.1.0.14100 64-Bit

----------------------------------------

pylon Viewer info: 7.4.0.14100 64-Bit

----------------------------------------

Host system infos:

CPU architecture: x86_64

Kernel type: linux

Kernel version: 6.5.0-35-generic

OS type: ubuntu

OS version: 22.04

----------------------------------------

Screen(s):

Screen 0 scale factor: 1

Screen 0 resolution: 1920x1080

----------------------------------------

Window(s):

"pylon Viewer 64-Bit" on screen 0

----------------------------------------

Host network infos:

Host network adapter enp3s0:

   Name: enp3s0

   MAC address: 00:D8:61:E5:6C:D4

Host network adapter wlo1:

   Name: wlo1

   MAC address: 40:EC:99:EB:6B:4E

   IP address: 128.39.51.43

   Subnet mask: 255.255.255.128

Host network adapter docker0:

   Name: docker0

   MAC address: 02:42:5C:0C:49:A9

   IP address: 172.12.0.1

   Subnet mask: 255.255.255.0

---------------------------------------
Ros distro : Humble
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2024-06-13:
Hello @Typos91 
I committed some changes regarding the services related to the line selector, mode, and source. Can you test your code with the new version please?

Typos91, 2024-06-13:
Hello, 
It works perfectly now, thank you !

FrancoisPicardDTI, 2024-06-13:
Great!
For the sake of documentation:
* Line1 is always Input mode, Line2 is always Output mode, but the other lines can be both. This was not implemented like that.
* I added the UserOutput2 and UserOutput3 sources. New sources can be added if needed, the code is documented accordingly.

I am closing the issue.

closed 2024-06-13 by FrancoisPicardDTI