basler/pylon-ros-camera #226: Error connecting to Basler A2a2600-20gcPRO  REV: 12 through default.yaml

tui-abdul, 2024-06-13:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I want to specify user id via config_file, which in this case is default.yaml, I have tried to configure it using following ros2 command "startup_user_set:=Default" but i face the following error.

pylon_ros2_camera_wrapper-1] 1718269086.782112861 [basler.pylon.ros2.pylon_ros2_gige_ace2_camera] [ERROR] Error applying camera specific startup setting for GigE cameras: Parameter not found in CFloatParameter::GetMax. (No node attached.)
[pylon_ros2_camera_wrapper-1] 1718269086.782186581 [basler.pylon.ros2.pylon_ros2_camera_node] [ERROR] Error while applying the user-specified startup settings (e.g., mtu size for GigE, ...) to the camera!
[pylon_ros2_camera_wrapper-1] 1718269086.782263233 [basler.pylon.ros2.pylon_ros2_camera_node] [ERROR] Error when trying to init and register. Shutting down now.

This problem is already reported [https://github.com/basler/pylon-ros-camera/issues/66#issue-787006921]

### Hardware setup description

Ubuntu 22.04 Jammy, x86_64, 
[image]

### Runtime information

```shell
ROS distribution is Humble and and i have installed Pylon-ros-camera Humble branch.
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2024-07-23:
Hello @tui-abdul 
Thank you for reporting this issue. I am sorry I don't manage to reproduce it. Can you tell me a bit more about how you set the startup_user_set. Do you do it through the default.yaml file, the launch file itself og the launch command line arguments?
Beware that some parameters implemented by the driver, like for instance the parameter `startup_user_set`, can be set through 1. the `pylon_ros2_camera_wrapper/config/default.yaml` user parameter file, 2. the `pylon_ros2_camera.launch.py` driver launch file, and 3. the command line arguments of the launch command to start the driver. A parameter value set as an argument of the launch command to start the driver will overwrite the value set in the driver launch file itself, that will overwrite the value set in the user parameter file.

FrancoisPicardDTI, 2024-10-31:
I am closing the issue for now, as there's no activity.

closed 2024-10-31 by FrancoisPicardDTI