basler/pylon-ros-camera #227: BUG: ROS 1 (master branch) package.xml version is not updated

peci1, 2024-06-14:
### Issue description

package.xml on branch master says version is 0.17.1. However, the changelog says 0.17.3. And I see the changes described in changelog for 0.17.3 are actually present in the code. So it seems that you forgot to update the version in package.xml .

### Code example about how to reproduce the issue

N/A

### Hardware setup description

N/A

### Runtime information

```shell
ROS Noetic / Ubuntu 20.04
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2024-06-18:
@peci1 thank you. It is now fixed.

closed 2024-06-18 by FrancoisPicardDTI

peci1, 2024-06-18:
Thanks! I suggest using `catkin_generate_changelogs` and `catkin_prepare_release` in the future so that such problems do not occur again.