basler/pylon-ros-camera #229: BUG: Large memory leak for every Blaze image

bracoe, 2024-06-17:
### Issue description

We can only run the wrapper for about 12 minutes before a memory leak crashes everything. Valgrind pointed out the leak fixed by this commit in a pull request I made.

https://github.com/basler/pylon-ros-camera/pull/224/commits/6886b28f3ed5abba0c0d364bb0fe4e60cb9d6d57

### Code example about how to reproduce the issue

Follow instructions for running the wrapper and use a Blaze 101.

### Hardware setup description

Ubuntu 22.04 and a Blaze 101

### Runtime information

```shell
Humble
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

lindhoscaa, 2024-06-18:
Same issue here, will try out this commit and see if it helps.

FrancoisPicardDTI, 2024-06-18:
Hello @bracoe 
Thanks for pointing out the bug. Your fix is not working - try to visualize the depth map and the color depth map in rqt - but you can free the pointers after the messages have been set.
I am including the fix in the next version of the driver later today.

FrancoisPicardDTI, 2024-06-18:
Hello @bracoe and @lindhoscaa 
The changes have been pushed. Please have a try.

closed 2024-06-18 by FrancoisPicardDTI

lauri9, 2024-06-19:
Hi, I also encountered this bug in a long running process. Instead of explicitly calling `new` and  `free`, I thought it might be good to let the `cv::Mat` object allocate and manage the memory. I did these two modifications in `pylon_ros2_camera_blaze.hpp`:
```
// depth map
cv::Mat depth_map = cv::Mat(height, width, CV_16UC1);
this->calculateDepthMap(range_component, min_depth, max_depth, (uint16_t*)depth_map.data);
```
and
```
// depth map color
cv::Mat depth_map_color = cv::Mat(height, width, CV_8UC3);
this->calculateDepthMapColor(range_component, min_depth, max_depth, (BGR*)depth_map_color.data);
```