basler/pylon-ros-camera #234: BUG: Increasing memory allocation when using /grab_images_raw action

tilljensen, 2024-07-30:
### Issue description

I've tried using the /grab_images_raw action for image acquisition from an ACE USB 3.0 camera but within some minutes my system ran out of memory and froze. It seems like that with each goal sent to the wrapper, the wrappers memory usage increases.

After roughly 30 seconds of running the modified action client mentioned below, the resident size increased from 150M to 2164M. Each action causes an increase of roughly 24M, lining up with the image size of the camera in mono8.

### Code example about how to reproduce the issue

Run the wrapper by using the included launch file:

```
ros2 launch pylon_ros2_camera_wrapper pylon_ros2_camera.launch.py 
```

Use the following command to repeatedly send goals to the action server:

```
ros2 action send_goal /my_camera/pylon_ros2_camera_node/grab_images_raw pylon_ros2_camera_interfaces/action GrabImages "{exposure_given : true,exposure_times : [30],gain_given : false,gain_values  : [],gamma_given : false,gamma_values  : [],brightness_given : false,brightness_values  : [],exposure_auto : false,gain_auto : true}"
```

It is also possible to reproduce this issue a little easier using the [pylon_ros2_camera_wrapper/test/test_grab_images_action_client.cpp](https://github.com/basler/pylon-ros-camera/blob/1ffd6a3eebc136ece3549bf629443b6ecfc8a0e8/pylon_ros2_camera_wrapper/test/test_grab_images_action_client.cpp) node after some modifications, namely commenting out the following lines:

-  57
- 153 to 162
- 165

Now rebuild, source and run the node using the command:

```
ros2 run pylon_ros2_camera_wrapper test_grab_image_action_client 
```

Now observe the memory usage of the wrapper process.

### Hardware setup description

Basler a2A5328-15ucPRO USB3.0
Kubuntu 22.04

### Runtime information

```shell
ROS2 humble
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2024-08-23:
Hello @tilljensen 
Can you try the newest push on the humble branch and see if it solves your problem? How do you monitor the memory usage?

FrancoisPicardDTI, 2024-10-31:
I am closing the issue for now, as there's no activity.

closed 2024-10-31 by FrancoisPicardDTI