basler/pylon-ros-camera #238: BUG: PylonROS2Camera not ready - pylon 7.5.0 pylon-supplementary-package-for-blaze 1.6.0

uff1199, 2024-08-09:
### Issue description

I am not able to launch the ROS2 driver with ROS Humble on ubuntu 22.04 from the commit: https://github.com/basler/pylon-ros-camera/commit/1ffd6a3eebc136ece3549bf629443b6ecfc8a0e8

Instead it is running with commit: 29397f6c7b82541f9fd2c43478c00938e4860eac

### Code example about how to reproduce the issue

 cd ~/dev_ws/src && git clone -b humble https://github.com/basler/pylon-ros-camera pylon_ros2_camera -b humble
cd pylon_ros2_camera

git checkout 1ffd6a3eebc136ece3549bf629443b6ecfc8a0e8

cd ~/dev_ws/src/pylon_ros2_camera && git clone https://github.com/ros-perception/image_common.git -b humble

cd ~/dev_ws && rosdep install --from-paths src --ignore-src -r -y

cd ~/dev_ws && colcon build

cd ~/dev_ws && . install/setup.bash

ros2 launch pylon_ros2_camera_wrapper pylon_ros2_camera.launch.py

### Error message

```shell
The error log is:
[INFO] [launch]: All log files can be found below /home/sensorint/.ros/log/2024-08-09-13-01-42-616784-sensorint-B360N-WIFI-90511        
[INFO] [launch]: Default logging verbosity is set to INFO                                                              
[INFO] [pylon_ros2_camera_wrapper-1]: process started with pid [90513]                                                                                                                                                                        
[pylon_ros2_camera_wrapper-1] 1723201302.825803759 [basler.pylon.ros2.pylon_ros2_camera_parameter] [INFO] Autoflash: 0, line2: 1, line3: 1                                                                          
[pylon_ros2_camera_wrapper-1] 1723201302.825915538 [basler.pylon.ros2.pylon_ros2_camera_parameter] [INFO] No Device User ID set -> Will connect the first available camera device
[pylon_ros2_camera_wrapper-1] 1723201304.641068441 [basler.pylon.ros2.pylon_ros2_camera] [INFO] Found camera device! Device Model: daA1280-54uc with Device User Id:                                                                           
[pylon_ros2_camera_wrapper-1] 1723201304.729931125 [basler.pylon.ros2.pylon_ros2_camera_node] [INFO] No user id for the camera has been set                                                                                                    
[pylon_ros2_camera_wrapper-1] 1723201304.794129123 [basler.pylon.ros2.pylon_ros2_usb_camera] [INFO] Startup user profile set to "CurrentSetting"                                                                                               
[pylon_ros2_camera_wrapper-1] 1723201304.794155911 [basler.pylon.ros2.pylon_ros2_usb_camera] [INFO] CurrentSetting loaded                                                                                                                     
[pylon_ros2_camera_wrapper-1] 1723201304.794185552 [basler.pylon.ros2.pylon_ros2_camera_base] [WARN] No image encoding provided -> Will use 'mono8' or 'rgb8' as fallback                                                                      
[pylon_ros2_camera_wrapper-1] 1723201304.853255395 [basler.pylon.ros2.pylon_ros2_camera_base] [ERROR] An image grabbing exception occurred: The camera device does not support waiting for frame trigger ready.                                
[pylon_ros2_camera_wrapper-1] 1723201304.853349113 [basler.pylon.ros2.pylon_ros2_camera_base] [ERROR] PylonROS2Camera not ready because the result of the initial grab is invalid                                                              
[pylon_ros2_camera_wrapper-1] 1723201304.868168381 [basler.pylon.ros2.pylon_ros2_camera_node] [WARN] [] name not valid for camera_info_manager                                                                                                 
[pylon_ros2_camera_wrapper-1] 1723201304.869630219 [basler.pylon.ros2.pylon_ros2_camera_node] [INFO] CameraInfoURL needed for rectification! ROS2-Param: '/my_camera/camera_info_url' = '' is invalid!                                         
[pylon_ros2_camera_wrapper-1] 1723201304.869717159 [basler.pylon.ros2.pylon_ros2_camera_node] [DEBUG] CameraInfoURL should have following style: 'file:///full/path/to/local/file.yaml' or 'package://<package_name>/relative/path/to/file.yaml'
[pylon_ros2_camera_wrapper-1] 1723201304.869767302 [basler.pylon.ros2.pylon_ros2_camera_node] [WARN] Will only provide distorted /image_raw images!                                                                                            
[pylon_ros2_camera_wrapper-1] 1723201304.919849858 [basler.pylon.ros2.pylon_ros2_camera_node] [INFO] Setting horizontal binning_x to 1                                                                                                        
[pylon_ros2_camera_wrapper-1] 1723201304.969289749 [basler.pylon.ros2.pylon_ros2_camera_node] [INFO] Setting vertical binning_y to 1    
[pylon_ros2_camera_wrapper-1] 1723201304.971633140 [basler.pylon.ros2.pylon_ros2_camera_node] [INFO] Startup settings: encoding = 'rgb8', binning = [1, 1], exposure = 1471, gain = 0, gamma = 1, shutter mode = default_shutter_mode          
[pylon_ros2_camera_wrapper-1] 1723201304.972052447 [basler.pylon.ros2.pylon_ros2_camera_node] [INFO] Start image grabbing if node connects to topic with a spinning rate of: 5 Hz                                                              
[pylon_ros2_camera_wrapper-1] 1723201305.172466032 [basler.pylon.ros2.pylon_ros2_camera_node] [INFO] Camera not calibrated                                                                                                                    
[pylon_ros2_camera_wrapper-1] 1723201305.173972781 [basler.pylon.ros2.pylon_ros2_camera_node] [WARN] Error in grabbingStopping(): pylon_camera_ is not ready!                                                                                  
[pylon_ros2_camera_wrapper-1] 1723201305.174026570 [basler.pylon.ros2.pylon_ros2_camera_node] [WARN] Error in grabbingStarting(): pylon_camera_ is not ready!                                                                                  
[pylon_ros2_camera_wrapper-1] 1723201305.174068065 [basler.pylon.ros2.pylon_ros2_camera_node] [WARN] Error in publishCurrentParams(): pylon_camera_ is not ready!                                                                              
[pylon_ros2_camera_wrapper-1] 1723201305.373726799 [basler.pylon.ros2.pylon_ros2_camera_node] [WARN] Error in publishCurrentParams(): pylon_camera_ is not ready!                                                                              
[pylon_ros2_camera_wrapper-1] 1723201305.573676753 [basler.pylon.ros2.pylon_ros2_camera_node] [WARN] Error in publishCurrentParams(): pylon_camera_ is not ready!                                                                              
[pylon_ros2_camera_wrapper-1] 1723201305.773673663 [basler.pylon.ros2.pylon_ros2_camera_node] [WARN] Error in publishCurrentParams(): pylon_camera_ is not ready!                                                                              
[pylon_ros2_camera_wrapper-1] 1723201305.973659495 [basler.pylon.ros2.pylon_ros2_camera_node] [WARN] Error in publishCurrentParams(): pylon_camera_ is not ready!                                                                              
[pylon_ros2_camera_wrapper-1] 1723201306.173687263 [basler.pylon.ros2.pylon_ros2_camera_node] [WARN] Error in publishCurrentParams(): pylon_camera_ is not ready!                                                                              
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
```

### Hardware setup description

Basler USB3 camera: daA1280-54uc

x86_64
PRETTY_NAME="Ubuntu 22.04.4 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.4 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

### Runtime information

```shell
ROS2 Humble
pylon/now 7.5.0.15658-deb0 amd64 [installed,local]
  Basler pylon Camera Software Suite

pylon-supplementary-package-for-blaze/now 1.6.0 amd64 [installed,local]
  Tools, GenTL producer, and samples for operating Basler blaze cameras.
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2024-08-15:
Hello @uff1199 
This is not due to the commit as there were only changes made in the documentation. It is more likely a problem coming from your camera, because of that:
`[pylon_ros2_camera_wrapper-1] 1723201304.853255395 [basler.pylon.ros2.pylon_ros2_camera_base] [ERROR] An image grabbing exception occurred: The camera device does not support waiting for frame trigger ready.
`
I'll get in touch about that. In the mean time, do you have another camera, maybe more recent, to test your code with?

uff1199, 2024-08-19:
Hi @FrancoisPicardDTI,

as mentioned in the description the camera is operational with the ROS2 driver built on [this commit]( commit https://github.com/basler/pylon-ros-camera/commit/29397f6c7b82541f9fd2c43478c00938e4860eac). However, it's not running with the latest changes made to the humble branch in the repository.

Thank's a lot for your help!

FrancoisPicardDTI, 2024-08-23:
Hi @uff1199 
Should be fixed now. Try the newest release on the humble branch and let me now if it is working.
Apparently the WaitForTriggerReady() function is not supported by some devices now. I tried with a dart model and I had the same problem. Fortunately there's a CanWaitForTriggerReady() that can be used beforehand.

FrancoisPicardDTI, 2024-10-31:
The problem is considered as fixed. Please reopen if it is not the case.