basler/pylon-ros-camera #239: BUG: Some includes changed in ROS2 Jazzy

ndahn, 2024-08-22:
### Issue description

In ROS2 Jazzy, the includes for `cv_bridge` and `image_geometry` have changed from .h to .hpp files. The driver compiles fine once the files making these includes are updated, namely:
- pylon_ros2_camera_component/include/pylon_ros2_camera_node.hpp
- pylon_ros2_camera_component/include/internal/impl/pylon_ros2_camera_blaze.hpp

### Code example about how to reproduce the issue

Not applicable

### Hardware setup description

Not applicable

### Runtime information

```shell
Not applicable
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2024-08-23:
Hello @ndahn 
The driver has not been tested with Jazzy. An official version for Jazzy is going to be released soon.

FrancoisPicardDTI, 2024-10-01:
Hello @ndahn 
Please test the jazzy_beta branch. It should solve your issue. If not, please feel free to reopen this issue.