basler/pylon-ros-camera #242: Unable to Decompress Images in ROS2 using Pylon SDK in Python (or without Pylon SDK)

ibr5456s, 2024-09-09:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hello,  

I am currently working on a project where I need to decompress images coming from a Basler camera in a ROS2 environment using the Pylon SDK in Python (or any other method that can achieve this goal) to visualize them in RVIZ2. The images are published to a ROS2 topic as `sensor_msgs/Image` messages, but they are actually compressed and appear as static noise when visualized in RVIZ2. My objective is to decompress these images and publish them again in a decompressed format that can be visualized correctly in RVIZ2.

### Issue and Steps to Reproduce  

I am using the following Python code to attempt to decompress the images using the Pylon SDK (still experminting):

```python
#!/usr/bin/env python3

import rclpy
from rclpy.node import Node
from sensor_msgs.msg import Image
from pypylon import pylon
from cv_bridge import CvBridge
import numpy as np
import cv2

class PylonImageDecompressorNode(Node):
    def __init__(self):
        super().__init__('pylon_image_decompressor')

        self.bridge = CvBridge()

        # Set up the Pylon image decompressor
        self.decompressor = pylon.ImageDecompressor()

        # Subscriber to the compressed image topic (published as Image)
        self.image_sub = self.create_subscription(
            Image,  # Message type for the incoming image even if it is compressed
            '/my_camera/pylon_ros2_camera_node/image_raw',  # The actual topic with compressed data
            self.image_callback,
            10
        )

        # Publisher to output the decompressed image
        self.image_pub = self.create_publisher(
            Image,  # Message type: Image
            'pylon/camera/decompressed_image',  # Topic to publish to
            10
        )

    def image_callback(self, msg):
        self.get_logger().info('Received a compressed image wrapped in an Image message.')

        try:
            # Extract raw data from the ROS2 Image message
            np_arr = np.frombuffer(msg.data, np.uint8)

            # Calculate buffer size in bytes
            buffer_size = np_arr.size * np_arr.itemsize

            # Create a Pylon image object
            pylon_image = pylon.PylonImage()

            # Attach the raw buffer to the Pylon image object using the correct parameters
            pylon_image.AttachUserBuffer(
                np_arr.ctypes.data,  # Pointer to the data
                buffer_size,  # Size of the data in bytes
                pylon.PixelType_Mono8,  # EPixelType for mono8 format
                msg.width,  # Image width
                msg.height,  # Image height
                0,  # PaddingX (no padding in mono8 format)
                pylon.ImageOrientation_TopDown  # Image orientation (default TopDown)
            )

            # Decompress the image using the Pylon SDK
[... 223 lines cut ...]
    Name: enp45s0
    MAC address: 58:11:22:89:3B:E1

Host network adapter docker0:
    Name: docker0
    MAC address: 02:42:9C:42:4B:72
    IP address: 172.17.0.1
    Subnet mask: 255.255.0.0

Host network adapter br-a296f01552f4:
    Name: br-a296f01552f4
    MAC address: 02:42:36:9B:A9:47
    IP address: 172.18.0.1
    Subnet mask: 255.255.0.0

Host network adapter wlp46s0:
    Name: wlp46s0
    MAC address: 14:13:33:88:A3:99
    IP address: 192.168.0.212
    Subnet mask: 255.255.255.0
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2024-09-17:
Hello @ibr5456s 
Thank you for your reporting.
You write: "they are actually compressed and appear as static noise when visualized in RVIZ2". I don't have your camera model but it should be possible to visualize the images in rviz and rqt. Did you try rqt?
I can't help you here with the Pylon decompressor, you should contact the Basler support to get some help from them.
Do your program work with images acquired from the pylon viewer and stored on your computer?

FrancoisPicardDTI, 2024-10-31:
I am closing the issue for now, as there's no activity.

closed 2024-10-31 by FrancoisPicardDTI