basler/pylon-ros-camera #245: A GigE camera's temperature is always 0.0

boxanm, 2024-09-20:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I noticed that the `temperature` field of the `/basler/current_params` topic is always 0.0.
After some investigation, I noticed that the [PR](https://github.com/basler/pylon-ros-camera/pull/50) that introduced temperature readings treats GigE and USB cameras differently.
Swapping the original code in the GigE camera call:
```C++
return static_cast<float>(cam_->TemperatureAbs.GetValue());
```
for the USB camera version:
```C++
return static_cast<float>(cam_->DeviceTemperature.GetValue());
```
fixed the issue, and I'm now able to monitor the camera's temperature. I'll be happy to submit a PR with this fix, assuming that the temperature calls in the Pylon SDK have changed since the original PR was submitted. Can someone with more experience with Pylon confirm my hypothesis?

### Hardware setup description

We have a Basler a2A1920-51gcPRO camera. I wasn't able to verify any other device.

### Runtime information

```shell
Ros2 Humble
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

SMA2016a, 2024-09-22:
Check if your camera has got the temperature sensor with pylonviewer.

boxanm, 2024-09-24:
Yes, I can get the temperature in PylonViewer, as well as when when using the `cam_->DeviceTemperature.GetValue()` call. Nevertheless, the code sample uses another function call to read it
[image]

closed 2025-02-18 by FrancoisPicardDTI