basler/pylon-ros-camera #248: Couldn't find the camera that matches the specified Device User ID but was found by pylonviewer

SebGoCCC, 2024-10-28:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I have three GiGE cameras (left, middle, right) connected to a Jetson Orin AGX via a [Forge Carrier Board](https://connecttech.com/product/forge-carrier-for-nvidia-jetson-agx-orin/). Using the pylon_camera_wrapper, I can only access two cameras and retrieve images (middle and right). It cannot find the camera with DeviceUserId=left. 
I can access all cameras through the Pylon Viewer without problems.

```
...
[pylon_ros2_camera_wrapper-1] 0000005504.003967552 [basler.pylon.ros2.pylon_ros2_camera_parameter] [INFO] Trying to connect the camera device with the following device user id: left
[pylon_ros2_camera_wrapper-1] 0000005505.809995008 [basler.pylon.ros2.pylon_ros2_camera] [ERROR] Couldn't find the camera that matches the specified Device User ID: left! Either the ID is wrong or the camera device is not connected (yet)
...
```

[image]

Does anyone have an idea what could be causing this?

### Hardware setup description

- Jetson ORIN AGX on a [Forge Carrier Board](https://connecttech.com/product/forge-carrier-for-nvidia-jetson-agx-orin/)
- Cameras: 3x a2A4096-9gcBAS
- OS Architecture: Linux 5.15.136-tegra

### Runtime information

```shell
pylon Release: 8.0.0.16021
pylon C++ SDK 9.0.3.215
pylon Application 3.3.0.16064
pylon Viewer 8.1.1.16064

ROS2 Distro: Humble
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2024-10-31:
Hello @SebGoCCC 
Does your left camera have a manually set IP address? See here: https://github.com/basler/pylon-ros-camera?tab=readme-ov-file#no-connection-with-connected-camera
Did you try to connect to just this one, by disconnecting from your network the 2 other cameras?

SebGoCCC, 2024-11-04:
Hi
Yes, all cameras have a manually assigned IP addresses. 
I disconnected the other cameras and tried again. It seems to be working now.
Do you have any idea what might have caused the issue so that I can avoid it in the future?

FrancoisPicardDTI, 2024-12-12:
Hello @SebGoCCC 
No I am sorry. It is difficult to evaluate what the problem could be, as I don't have your setup.
Anyway, great that it worked out at the end!
I am closing the issue.

closed 2024-12-12 by FrancoisPicardDTI