basler/pylon-ros-camera #252: Error Applying Camera Specific Startup Settings with acA640 USB Camera on Ubuntu 22 and ROS Humble

neerja123-gupta, 2024-11-04:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I am experiencing an issue when trying to launch the Basler Pylon ROS2 camera using the pylon_ros2_camera_wrapper. The camera is recognized, and I can visualize the image using the Pylon Viewer (version 7.5.0). However, when attempting to run the camera node via ROS2, 

Launch command
ros2 launch pylon_ros2_camera_wrapper pylon_ros2_camera.launch.py

I encounter several errors.

### Hardware setup description

System Information:

    OS: Ubuntu 22.04
    ROS Version: Humble
    Pylon Viewer Version: 7.5.0
    Camera Model: Basler acA640-750uc

### Runtime information

```shell
Log Output:
![Screenshot from 2024-11-04 14-29-54](https://github.com/user-attachments/assets/3e2002aa-f9fe-46c4-81a2-9f1237e3f3e4)

Steps to Reproduce:

    Install the Basler Pylon SDK and ROS2 packages.
    Connect the Basler acA640-750uc USB camera.
    Launch the camera using the command above.

Expected Behavior: The camera should initialize correctly and start streaming images via ROS2.

Actual Behavior: The process fails with errors related to startup settings and unable to spin.

Additional Information:

    I can visualize the image in the Pylon Viewer without any issues.
    Please let me know if additional logs or configurations are needed for troubleshooting.

Thank you for your assistance!
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

neerja123-gupta, 2024-11-05:
Solved by setting the startup_user_set param in launch file.

closed 2024-11-05 by neerja123-gupta

roboticshard, 2024-11-06:
Hi @neerja123-gupta ,
Could you provide me more information how did you solve it ? I am having same issue. In addition to that issue, my USB 3.0 Basler camera in Pylon Viewer is working perfectly in terms of high FPS, color without any parameter settings as a initial. For ROS2 Humble, I need to optimize this parameters by myself manuelly. Otherwise, I acquire low FPS, different color something like that.

neerja123-gupta, 2024-11-06:
Hi, for my setup, I ended up commenting out all the user-specific startup settings because the default configurations weren’t working for my camera. I commented out lines 30, 61, 102–107, and 135 in the pylon_ros2_camera.launch.py file from [the pylon_ros2_camera GitHub repository](https://github.com/basler/pylon-ros-camera/blob/humble/pylon_ros2_camera_wrapper/launch/pylon_ros2_camera.launch.py). After making these changes, I was able to view images in RViz2. However, I'm also encountering an issue in RViz (Humble): it’s displaying monochrome (mono8) images, even though I can see proper color images in the Pylon Viewer.

roboticshard, 2024-11-07:
Hi @neerja123-gupta I was having same problem with mono and rgb that is why I recommend try to image_encoding: "bayer_gbrg8" in default.yml. It worked for me but in my case the fps is decreased . I can see proper color images in Pylon Viewer but in rqt /my_camera/pylon_ros2_camera_node/image_raw/ viewing slowly with low fps. I have already set frame_rate and exposure time with same parameter in pylon. I dont know for which reason does not work. Could you guide me how can I view in Rviz2?

neerja123-gupta, 2024-11-07:
Hi @roboticshard, yes, it’s working for me now too. Previously, I was modifying the parameters through the command line, but after updating them in the YAML file, I can now see the rgb8 images. To view the images in RViz, just open RViz2 in a separate terminal, click on "Add," select "Topic," or simply add an "Image" topic manually and choose the appropriate topic name. You should then be able to see the image.

roboticshard, 2024-11-07:
@neerja123-gupta  I cannot view the image in ROS2 as fast as in the Pylon Viewer. I think there is some delay. Although I am settings many camera parameters in ROS2 as same as in the Pylon Viewer. Do you have any suggestion? I realized that/ my_camera/pylon_ros2_camera_node/image_raw/theora is a little bit faster compared to /my_camera/pylon_ros2_camera_node/image_raw. ROS2 frame rate is 5 Hz, however Pylon frame rate is 15 Hz.