basler/pylon-ros-camera #255: Basler Dart M Camera with ROS2

Pandananana, 2024-11-14:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I have a Dart-M camera (DMA 1920-51GC-LS). Is this not supported by this ROS2 driver?

I am getting this error when running the launch command: ros2 launch pylon_ros2_camera_wrapper pylon_ros2_camera.launch.py

```
[pylon_ros2_camera_wrapper-1] 1731576632.307847230 [basler.pylon.ros2.pylon_ros2_camera] [ERROR] Found 'BaslerGigE' camera device type, but it is neither a ace, nor a ace2 camera device. Other camera types are not supported by this driver for now!
```

The error message comes form pylon_ros2_camera.cpp file.
Is there another branch that supports my camera, or is it really only ace and ace2 cameras that are supported by this driver?

### Hardware setup description

Camera Serial: 40556314
Firmware Version: p=mg2_imx392c/s=r/v=1.0.1/i=9122.11/h=8170e60
CPU: x86
OS: Ubuntu 22.04

### Runtime information

```shell
Could not get Shift + Ctrl + Alt + i to work

ROS Distro: Humble
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2024-11-27:
Hello @Pandananana 
The Dart-M model is not supported (yet) by the ROS2 driver. It is not planned for now but I am keeping this issue opened in case of several users ask for this support.
In any case, you more than welcome to suggest a pull request if you integrate this model in the existing pylon driver.

SMA2016a, 2025-01-31:
@Pandananana, you may easily work around this if you set fooling in the cpp. pylon_ros2_camera.cpp in Line 82. I did not test it, but it must work.

else if (model_name.compare(0, 3, "a2A") == 0 || model_name.compare(0, 3, "dmA") == 0)

 {

         return GIGE2;

  }

sw010129, 2025-03-12:
Modified line 82 of the pylon_ros2_camera.cpp file as you did and the problem persists, what should I do

epiphyte, 2025-08-01:
You should give https://github.com/magazino/pylon_camera a try.