basler/pylon-ros-camera #257: Basler Ace2 GigECamera can't lauch pylon_ros2_camera_wrapper

leiwanrobotics, 2024-11-29:
### Issue description

Basler Ace2 GigE camera
Pylon 7.2
Pylon-supplementary 1.3
ROS2 Humble
Ubuntu 22.04

Pylonviewer can successfuly launch the camera and get images, but when I run ros2 launch, it always show the error: 

Error applying camera specific startup setting for GigE cameras: Parameter not found in CFloatParameter::GetMax. (No node attached.)

### Code example about how to reproduce the issue

My default yaml:

# Camera parameters

/**:
  ros__parameters:

    #  The tf frame under which the images were published
    camera_frame: basler_camera

    #  The DeviceUserID of the camera. If empty, the first camera found in the
    #  device list will be used
    device_user_id: "40522320"

    #  The CameraInfo URL (Uniform Resource Locator) where the optional intrinsic
    #  camera calibration parameters are stored. This URL string will be parsed
    #  from the ROS-CameraInfoManager:
    camera_info_url: ""

    #  The encoding of the pixels -- channel meaning, ordering, size
    #  taken from the list of strings in include/sensor_msgs/image_encodings.h
    #  The supported encodings are 'mono8', 'bgr8', 'rgb8', 'bayer_bggr8',
    #  'bayer_gbrg8' and 'bayer_rggb8'
    #  Default values are 'mono8' and 'rgb8'
    image_encoding: "bayer_rggb8"

    #  Binning factor to get downsampled images. It refers here to any camera
    #  setting which combines rectangular neighborhoods of pixels into larger
    #  "super-pixels." It reduces the resolution of the output image to
    #  (width / binning_x) x (height / binning_y).
    #  The default values binning_x = binning_y = 0 are considered the same
    #  as binning_x = binning_y = 1 (no subsampling).
    # binning_x: 1
    # binning_y: 1

    #  The desired publisher frame rate if listening to the topics.
    #  This parameter can only be set once at startup
    #  Calling the GrabImages-Action can result in a higher framerate
    frame_rate: 30.0

    #  Mode of camera's shutter.
    #  The supported modes are "rolling", "global" and "global_reset"
    #  Default value is "" (empty) means default_shutter_mode
    # shutter_mode: ""

    ##########################################################################
    ######################## Image Intensity Settings ########################
    ##########################################################################
    # The following settings do *NOT* have to be set. Each camera has default
    # values which provide an automatic image adjustment resulting in valid
    # images
    ##########################################################################

    #  The exposure time in microseconds to be set after opening the camera.
    # exposure: 1000.0

[... 92 lines cut ...]
[pylon_ros2_camera_wrapper-1] 1732878125.213153122 [basler.pylon.ros2.pylon_ros2_camera_node] [ERROR] Error while applying the user-specified startup settings (e.g., mtu size for GigE, ...) to the camera!
[pylon_ros2_camera_wrapper-1] 1732878125.213396284 [basler.pylon.ros2.pylon_ros2_camera_node] [ERROR] Error when trying to init and register. Shutting down now.
[pylon_ros2_camera_wrapper-1] Impossible to spin
[pylon_ros2_camera_wrapper-1] failed to create guard condition: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at ./src/rcl/guard_condition.c:67
[ERROR] [pylon_ros2_camera_wrapper-1]: process has died [pid 9530, exit code 1, cmd '/home/xitaso/ros2_ws/install/pylon_ros2_camera_wrapper/lib/pylon_ros2_camera_wrapper/pylon_ros2_camera_wrapper --ros-args -r __node:=pylon_ros2_camera_node -r __ns:=/my_camera --params-file /home/xitaso/ros2_ws/install/pylon_ros2_camera_wrapper/share/pylon_ros2_camera_wrapper/config/default.yaml --params-file /tmp/launch_params_ujiir7sa'].
```

### Hardware setup description

CPU architecture X86_64
Ubuntu 22.04

### Runtime information

```shell
Basler Ace2 GigE camera
Pylon 7.2
Pylon-supplementary 1.3
ROS2 Humble
Ubuntu 22.04
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

tomaszjezak, 2024-12-04:
Basler made some change to this driver last month, it causes this error on my end too. Use an older version and it'll work.

leiwanrobotics, 2024-12-05:
> Basler made some change to this driver last month, it causes this error on my end too. Use an older version and it'll work.

Thanks for your response. Which older version works for you?

TaqiHamoda, 2024-12-10:
I have found a hacky solution that works but it is just a temporary fix for now.

I am using `CurrentSetting` as my startup user profile. In the `pylon_ros2_camera_component/include/internal/impl/pylon_ros2_camera_gige_ace2.hpp` , I simply removed the section that deals with that user profile and just let the code default to the `else` statement.

Here is the diff:
```diff
*** pylon_ros2_camera_gige_ace2.hpp	2024-12-10 07:16:11.794699946 +0100
--- pylon_ros2_camera_gige_ace2.hpp	2024-12-10 07:19:52.972429588 +0100
***************
*** 238,264 ****
  
              RCLCPP_INFO(LOGGER_GIGE_ACE2, "UserSet3 loaded");
          }
-         else if (parameters.startup_user_set_ == "CurrentSetting")
-         {
-             /* Thresholds for the AutoExposure Functions:
-                 *  - lower limit can be used to get rid of changing light conditions
-                 *    due to 50Hz lamps (-> 20ms cycle duration)
-                 *  - upper limit is to prevent motion blur
-                 */
-             double upper_lim = std::min(parameters.auto_exposure_upper_limit_, cam_->ExposureTimeAbs.GetMax());
-             cam_->AutoExposureTimeAbsLowerLimit.SetValue(cam_->ExposureTimeAbs.GetMin());
-             cam_->AutoExposureTimeAbsUpperLimit.SetValue(upper_lim);
-             RCLCPP_INFO_STREAM(LOGGER_GIGE_ACE2, "Cam has upper exposure value limit range: ["
-                     << cam_->ExposureTimeAbs.GetMin()
-                     << " - " << upper_lim << " (max possible value from cam is " << cam_->ExposureTimeAbs.GetMax() << ")"
-                     << "].");
-                     
-             cam_->GevSCPSPacketSize.SetValue(parameters.mtu_size_);
-             cam_->GevSCPD.SetValue(parameters.inter_pkg_delay_);
-             cam_->GevSCFTD.SetValue(parameters.frame_transmission_delay_);
- 
-             RCLCPP_INFO(LOGGER_GIGE_ACE2, "CurrentSetting loaded");
-         }
          else
          {
              RCLCPP_WARN_STREAM(LOGGER_GIGE_ACE2, "Unsupported startup user profile \"" << parameters.startup_user_set_ << "\", ignoring");
--- 238,243 ----
```

This is definitely just a band-aid fix until the deeper issue gets resolved. By playing a bit with the code and examining it, it seems that the newly added auto exposure functionality wasn't properly set up in the camera. Even when I replaced `cam_->ExposureTimeAbs` with a hard-coded value, `cam_->AutoExposureTimeAbsLowerLimit.SetValue` would just throw an error about  "No Node attached" like in the error message above.

I wanted to investigate this a bit more and create a PR but setting up the debugger would take a while and I am currently a bit busy with work/school :/

FrancoisPicardDTI, 2024-12-12:
Hi all
Can you please test the newest release of the humble branch ? Let me know if your problem has been fixed.

besma14, 2024-12-12:
Hello, 
I had the same problem, I just had to unplug my GigE camera and plug it back in, that's all.
on the other hand please tell me how you were able to wait for 30fps?

TaqiHamoda, 2024-12-18:
> Hi all
> Can you please test the newest release of the humble branch ? Let me know if your problem has been fixed.

Hello,

Yes the issue has been fixed now. It all works perfectly, thank you.

closed 2024-12-18 by FrancoisPicardDTI