basler/pylon-ros-camera #259: Resize and encoding image and FPS

besma14, 2024-12-11:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hello,

I am using a Basler GigE acA1920-50gc camera on Ubuntu 22.04 with ROS Humble, and I am encountering several issues that I cannot resolve:

    1.Grayscale image (mono8): My video stream is always displayed in grayscale, but I want it to be in RGB. I have tried modifying the parameters in the configuration file, but these changes are not applied. Even when forcing the parameters in the launch file, it still doesn’t work.

   2. Image resizing: Currently, the captured image is in Full HD (1920x1200). I want to resize it to 480x640. However, when I try to do this with OpenCV, the image becomes unusable: it is highly pixelated and distorted. Is there an effective method to perform this resizing?

    3.FPS limitation: I am limited to a maximum of 20 FPS, which is insufficient for my application. Is there a way to increase the frame rate?

I would greatly appreciate it if you could help me solve these issues.

Thank you in advance for your support.

### Hardware setup description

ROS2 humble 
ubuntu 22.04

### Runtime information

```shell
yes
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2024-12-12:
Hello @besma14 

1. Did you try other encodings than 'rgb8' ? As specified in the parameter file, the following encordings by the driver are : 'mono8', 'bgr8', 'rgb8', 'bayer_bggr8', 'bayer_gbrg8' and 'bayer_rggb8'. Otherwise, it not supported by the driver.
2. This is an OpenCV problem, I suggest looking at the OpenCV documentation.
3. There's a troubleshooting section regarding the framerate in the documentation here: https://github.com/basler/pylon-ros-camera?tab=readme-ov-file#slow-frame-rate. Have a look at the different solutions brought by other users. This is a known problem, still to be solved.

besma14, 2024-12-12:
Hello @FrancoisPicardDTI 
First of all, thank you for your answer. I did, however, encounter a few difficulties that I would like to share, hoping that you could help me or consider future improvements.
The only encoding that allows me to get a color image is 'bayer_rggb8', but when I set the encoding setting to rbg8, I get a grayscale image. Is there a way to correct this or get color images in RGB format directly?
 
I'm looking to resize images from 1200x1920 to 480x640 without using OpenCV, because all the solutions I've tested so far result in an image that is too pixelated. Would there be a native method or best practice to perform this resizing efficiently?
 
I've explored all the solutions mentioned in the similar issues and links provided in your documentation and the GitHub repository, but unfortunately, none of them have solved these issues.

FrancoisPicardDTI, 2025-02-18:
Hello @besma14 

rbg8 and bayer_rggb8 are different encodings so they can't just be assimilated like that. Nevertheless, if it is a problem on your side to set explicitely the 'bayer_rggb8' encoding, you can change the code locally, I guess. Setting the 'bayer_rggb8' allows you to get images in RGB format.

As for OpenCV, I would refer to the OpenCV community, I can't help you on this one, sorry.

closed 2025-09-30 by FrancoisPicardDTI