basler/pylon-ros-camera #261: BUG: UserSet is overwrite by launch file

diablo1281, 2025-01-25:
### Issue description

I tried to define UserSet in `YAML` file, but it seems that launch file overwrite it with `CurrentSettings`.

After commenting line on launch file on node construction, the setting from config file is loaded as it should. Without it, it always use `CurrentSettings` if not stated otherwise with launch command.

### Code example about how to reproduce the issue

None for now

### Error message

```shell

```

### Hardware setup description

Jetson Orin with ACE 2 USB camera

### Runtime information

```shell
Docker image with ROS2 Humble(Isaaac) with working pylon ROS driver.
```

### Is your camera operational with the Basler pylon Viewer on your platform?

Yes

FrancoisPicardDTI, 2025-02-18:
Hello @diablo1281 
Yes, the user set is set through the launch file. You can indeed set it through the launch command and its arguments. It is not  a bug.

diablo1281, 2025-05-21:
Hi @FrancoisPicardDTI, sorry for late response - I missed notification.
If I need to use separate launch files for 10 camera instances, it seems for me like a bug. As there need to be 10 launch files and 10 config files.